HiddenRamblings/TagMo #649: [Bug]: issues using N2 Elite with TagMo

Rowleyjp, 2022-12-11:
Have been experiencing increasingly more issues using the N2 Elite following the apps transition onto the Google Play Store.
Problems include not being able to change the N2 Elite bank count, trying to use the feature crashes the app.
Unable to add new amiibo to the N2 Elite within blank slots, trying to do so crashes the app.
I have attempted to use the erase all banks feature however this also crashes, after restarting the app there is 1 bank shown and the bank count can't be increased as it crashes the app.
I have also experienced general usability issues with the app since it has been on the Play Store, text not showing correctly, UI difficult to navigate when managing the N2 Elite and amiibo showing error messages after replacing existing banks on the N2 Elite.

AbandonedCart, 2022-12-11:
To quote [the post on SO](https://www.reddit.com/r/tagmo/comments/zi1yms/n2_elite_not_working_with_tagmo_any_more/)

> Have you tried reporting them through the app? It's the top right overflow menu (3 dots). After a crash, you go there and select "Report Problem"

AbandonedCart, 2022-12-11:
You also appear to be describing issues that happened from the initial conversion to Kotlin, which didn't start until 6 months after the initial Google Play release and were fixed a day later.

If you do feel that your issues are related to the app being on the Play Store, I should remind you that installing the Google Play version over GitHub is completely optional and not recommended. Google Play is only recommended for those unable to install through GitHub for whatever reason.

AbandonedCart, 2022-12-11:
Fixed by https://github.com/HiddenRamblings/TagMo/commit/7e49d08d4ccf415d53e582e73da84a43cf372fbf and https://github.com/HiddenRamblings/TagMo/commit/e47ff6293647661c46e1c3c213557e5d5245ad12

Still being tested further before submission to Google Play

closed 2022-12-11 by AbandonedCart

Rowleyjp, 2022-12-11:
Thanks, i'll try the new Google Play version once the update has been pushed through and submit the crash reports if I experience any further issues.