HiddenRamblings/TagMo #664: [Bug]:

nath30, 2023-03-10:
TagMo 3.6.5 (GitHub Release) #b115831
samsung O (8.0.0) - 3.00 GB RAM

--------- beginning of main
--------- beginning of crash
03-10 20:22:58.828 26202 26335 E AndroidRuntime: FATAL EXCEPTION: pool-1-thread-1
03-10 20:22:58.828 26202 26335 E AndroidRuntime: Process: com.hiddenramblings.tagmo.eightbit, PID: 26202
03-10 20:22:58.828 26202 26335 E AndroidRuntime: java.lang.RuntimeException: Can't create handler inside thread that has not called Looper.prepare()
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at android.os.Handler.<init>(Handler.java:203)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at android.os.Handler.<init>(Handler.java:117)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at android.content.ClipboardManager$2.<init>(ClipboardManager.java:84)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at android.content.ClipboardManager.<init>(ClipboardManager.java:84)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at android.app.SystemServiceRegistry$12.createService(SystemServiceRegistry.java:379)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at android.app.SystemServiceRegistry$12.createService(SystemServiceRegistry.java:377)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at android.app.SystemServiceRegistry$CachedServiceFetcher.getService(SystemServiceRegistry.java:1402)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at android.app.SystemServiceRegistry.getSystemService(SystemServiceRegistry.java:1354)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at android.app.ContextImpl.getSystemService(ContextImpl.java:1695)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at androidx.appcompat.view.ContextThemeWrapper.getSystemService(ContextThemeWrapper.java:180)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at android.view.ContextThemeWrapper.getSystemService(ContextThemeWrapper.java:171)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at android.app.Activity.getSystemService(Activity.java:6008)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at com.hiddenramblings.tagmo.eightbit.io.Debug.submitLogcat(Debug.kt:277)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at com.hiddenramblings.tagmo.eightbit.io.Debug.processLogcat(Debug.kt:334)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at com.hiddenramblings.tagmo.browser.BrowserActivity.onReportProblemClick$lambda$33(BrowserActivity.kt:793)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at com.hiddenramblings.tagmo.browser.BrowserActivity.$r8$lambda$om8Jfs2GkMnZBctC-sReFeEsBrc(BrowserActivity.kt:0)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at com.hiddenramblings.tagmo.browser.BrowserActivity$$ExternalSyntheticLambda17.run(R8$$SyntheticClass:0)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
03-10 20:22:58.828 26202 26335 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:764)

AbandonedCart, 2023-03-10:
Are you using files or document storage? For files, you would have accepted the all files permission. For document storage, you would have selected a starting folder.

nath30, 2023-03-10:
After some testing it will gladly serch folders on SD ("emulated") but not internal storage ("physical ").
But then unchecking the box dousn't stop it from searching sub folders

AbandonedCart, 2023-03-10:
Internal storage is not accessible to apps. Physical is a microsd card. Emulated is the built-in sdcard.

nath30, 2023-03-10:
Sorry guessing i got the labels backwards.
The old Tagmo worked fine on the phones storage befor but no problem moveing back to the SD.
But if im understanding corectly Foomiibo should just replace everything excluding my backup folder anyway.
If this is how the folder system workes now just close my issue. Thanks.

AbandonedCart, 2023-03-10:
WearOS is the watch version of Android. There is nowhere to insert an sdcard.

The old way to access files through an internal file manager, which most tutorials said to disable, only accepted a single root location for storage, which is /sdcard. Android has changed a lot since then. 

If you try to access storage by using /sdcard, it will fail because it is only a link. You have to access it from /storage/emulated/0. This now requires special permission. A few phones have a physical sdcard slot, but also have /storage/emulated/0 already linked to /sdcard, so the physical sdcard gets a serial number as an address. This means it is something like /storage/D786-HJGD7.  If you try to access the external storage, it may go to either address. That's typically luck of the draw. 

There is a whole other option where an sdcard gets converted into an extension of the emulated storage, but apps jsut see it as emulated, so that's not important. All of this confusion led to having an option to tell TagMo which one you were trying to access. Without the button, it would end up like most apps where TagMo gives you one folder you can place your files in and tells you where it has to be located.

closed 2023-04-07 by AbandonedCart