HiddenRamblings/TagMo #669: [Feature]: I have request can we use external RFID

cahyadiy1309, 2023-04-11:
### Feature Explanation

can I request use external RFID for my phone, because my phone don't have NFC. I buy RFID ACS ACR122U can u help me for used that? thank you 

### Usage Confirmation

- [X] I confirm this is a request for a new feature, not an issue

cahyadiy1309, 2023-04-11:
[image]
[image]
[image]

i attach picture that work in other app, can u help me? look in under reader and on left button list, that app can detect

AbandonedCart, 2023-04-11:
I am assuming the app that works is one that came with the device, right?

cahyadiy1309, 2023-04-12:
that app i new install, from (https://cloud.electron.id/s/F5SzkebenQBj3fw) in folder ACS-EVK-Android-115-A. that driver in use for that tools (RFID). my phone just normal cellphone. no application installed like that. are u have suggest?

cahyadiy1309, 2023-04-12:
import android.app.PendingIntent;
import android.content.BroadcastReceiver;
import android.content.Context;
import android.content.Intent;
import android.content.IntentFilter;
import android.hardware.usb.UsbDevice;
import android.hardware.usb.UsbManager;
import android.os.Bundle;
import android.support.v7.app.AppCompatActivity;
import android.widget.Toast;

public class MainActivity extends AppCompatActivity {

    private static final String ACTION_USB_PERMISSION = "com.example.USB_PERMISSION";
    private UsbManager usbManager;
    private PendingIntent permissionIntent;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);

        usbManager = (UsbManager) getSystemService(Context.USB_SERVICE);
        permissionIntent = PendingIntent.getBroadcast(this, 0, new Intent(ACTION_USB_PERMISSION), 0);

        // Register BroadcastReceiver to request USB permission
        IntentFilter filter = new IntentFilter(ACTION_USB_PERMISSION);
        registerReceiver(usbReceiver, filter);

        // Check for connected USB devices
        checkConnectedDevices();
    }

    @Override
    protected void onDestroy() {
        super.onDestroy();
        unregisterReceiver(usbReceiver);
    }

    private void checkConnectedDevices() {
        UsbDevice device = null;
        // Iterate through all connected USB devices
        for (UsbDevice usbDevice : usbManager.getDeviceList().values()) {
            // Check if the USB device is an RFID reader
            if (isRfidDevice(usbDevice)) {
                device = usbDevice;
                break;
            }
        }
        if (device != null) {
            // Request permission to access the RFID reader
            usbManager.requestPermission(device, permissionIntent);
        } else {
            Toast.makeText(this, "RFID reader not found", Toast.LENGTH_SHORT).show();
        }
    }

    private boolean isRfidDevice(UsbDevice device) {
        // Implement your logic here to determine if the USB device is an RFID reader
        // For example, by checking the device's vendor and product IDs
        // You may need to refer to the documentation of the specific RFID reader for this information
        // Return true if the device is an RFID reader, false otherwise
        return false;
    }

    private final BroadcastReceiver usbReceiver = new BroadcastReceiver() {
        @Override
        public void onReceive(Context context, Intent intent) {
            String action = intent.getAction();
            if (ACTION_USB_PERMISSION.equals(action)) {
                synchronized (this) {
                    UsbDevice device = intent.getParcelableExtra(UsbManager.EXTRA_DEVICE);
                    if (intent.getBooleanExtra(UsbManager.EXTRA_PERMISSION_GRANTED, false)) {
                        // Permission granted, you can now read RFID data from the device
                        // Implement your logic here to read RFID data from the device
                        // You may need to refer to the documentation of the specific RFID reader for reading data
                    } else {
                        Toast.makeText(context, "Permission denied for USB device", Toast.LENGTH_SHORT).show();
                    }
                }
            }
        }
    };
}

this example code for android but need adjustment for code, this code android can read/detect external RFID, can u implement that?

AbandonedCart, 2023-04-12:
It's not that simple.

cahyadiy1309, 2023-04-12:
oke, but i will wait if u can implement that, u have reference for windows for same like tagmo? thx

AbandonedCart, 2023-04-12:
https://www.acs.com.hk/en/mobile-support/ is the same library and there is some discussion about trying to use it in place of NFC hardware. The problem is that you are trying to use it on a device that does **not** have NFC. 

The library allows the reader to act in place of the internal NFC hardware, but the NFC commands are still performed by the **phone** NFC APIs. A phone without NFC lacks those APIs. 

Basic read and write are made possible by the library, but it would require creating a full replica of the internal APIs to turn that into something that would work with TagMo.

AbandonedCart, 2023-04-12:
If there are 20 people that own an ACR122U and their only phone does not have NFC, then I will try to add support. The reason for requiring 20 people is the amount of work required. If there are only one or two people that will use it, then an alternative seems like a much better option.

cahyadiy1309, 2023-04-12:
oky, thx u for explanation this problem, i suggest can u make vote for this Feature who don't have ncf in phone / have nfc by phone for data who want this update. thx u

AbandonedCart, 2023-04-13:
I did post a poll and it was shared in one of the more common amiibo communities, but there has not been a single vote from anyone that both had this hardware and didn't already have NFC on their phone. The poll still has a day left, though.

cahyadiy1309, 2023-04-13:
oke thx u, i will close this case. i hope this feature u can implement. thx for your response