HiddenRamblings/TagMo #672: [Issue]: n2 elite operations stuck at Mifare Ultralight scanning

ludeviance, 2023-05-01:
### Issue Description

Every operation I try to make on the n2 elite tab (write to one or many slots, set active slot, delete slot) get stuck at the tag reading screen with the message "mifare ultralight scanning..."

[image]

If I find an amiibo image from the browser tag and write from there, after I select the slot number and place the n2 behind my phone, I get the same screen, but with the message "N2 elite scanning..." and after less than a second, the operation is completed successfully. That way, I can only write on one slot and nothing more. 

### Logcat Output

```shell
improperly called by android.widget.LinearLayout{25d3b4d V.E..V... ........ 0,0-1048,217} during second layout pass: posting in next frame
05-01 11:49:49.362 15587 15587 W View    : requestLayout() improperly called by android.widget.LinearLayout{5543486 V.E..V... ........ 0,62-827,98 #7f090116 app:id/group1} during second layout pass: posting in next frame
05-01 11:49:49.362 15587 15587 W View    : requestLayout() improperly called by android.widget.LinearLayout{27f6147 V.E..V... ........ 0,98-827,134 #7f090117 app:id/group2} during second layout pass: posting in next frame
05-01 11:49:49.362 15587 15587 W View    : requestLayout() improperly called by android.widget.LinearLayout{6438705 V.E..V... ........ 0,0-1048,217} during second layout pass: posting in next frame
05-01 11:49:49.362 15587 15587 W View    : requestLayout() improperly called by android.widget.LinearLayout{3ca3d74 V.E..V... ........ 0,62-827,98 #7f090116 app:id/group1} during second layout pass: posting in next frame
05-01 11:49:49.362 15587 15587 W View    : requestLayout() improperly called by android.widget.LinearLayout{78db29d V.E..V... ........ 0,98-827,134 #7f090117 app:id/group2} during second layout pass: posting in next frame
05-01 11:49:49.362 15587 15587 W View    : requestLayout() improperly called by android.widget.LinearLayout{8c091bd V.E..V... ........ 0,0-1048,217} during second layout pass: posting in next frame
05-01 11:49:49.362 15587 15587 W View    : requestLayout() improperly called by android.widget.LinearLayout{62f1312 V.E..V... ........ 0,62-827,98 #7f090116 app:id/group1} during second layout pass: posting in next frame
05-01 11:49:49.362 15587 15587 W View    : requestLayout() improperly called by android.widget.LinearLayout{bc6cae3 V.E..V... ........ 0,98-827,134 #7f090117 app:id/group2} during second layout pass: posting in next frame
05-01 11:49:49.824 15587 15616 W Parcel  : Expecting binder but got null!
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2023-05-01:
What happened to the rest of the log?

ludeviance, 2023-05-01:
Sorry, I think I miss-copied the log. Here it is:

```shell
TagMo 3.7.5 (GitHub Release) #3fa2080
Xiaomi TIRAMISU (13) - 10.00 GB RAM

--------- beginning of system
05-01 11:49:11.279 15587 15642 W BpBinder: Slow Binder: BpBinder transact took 478 ms, interface=com.android.vending.billing.IInAppBillingService, code=9 oneway=false
05-01 11:49:11.280 15587 15646 W BpBinder: Slow Binder: BpBinder transact took 482 ms, interface=com.android.vending.billing.IInAppBillingService, code=9 oneway=false
05-01 11:49:11.571 15587 15642 W BpBinder: Slow Binder: BpBinder transact took 291 ms, interface=com.android.vending.billing.IInAppBillingService, code=9 oneway=false
05-01 11:49:11.700 15587 15647 W BpBinder: Slow Binder: BpBinder transact took 904 ms, interface=com.android.vending.billing.IInAppBillingService, code=9 oneway=false
05-01 11:49:11.980 15587 15645 W BpBinder: Slow Binder: BpBinder transact took 1179 ms, interface=com.android.vending.billing.IInAppBillingService, code=9 oneway=false
05-01 11:49:12.106 15587 15646 W BpBinder: Slow Binder: BpBinder transact took 825 ms, interface=com.android.vending.billing.IInAppBillingService, code=9 oneway=false
05-01 11:49:12.287 15587 15649 W BpBinder: Slow Binder: BpBinder transact took 1479 ms, interface=com.android.vending.billing.IInAppBillingService, code=9 oneway=false
05-01 11:49:12.362 15587 15642 W BpBinder: Slow Binder: BpBinder transact took 790 ms, interface=com.android.vending.billing.IInAppBillingService, code=9 oneway=false
05-01 11:49:12.605 15587 15643 W BpBinder: Slow Binder: BpBinder transact took 1796 ms, interface=com.android.vending.billing.IInAppBillingService, code=9 oneway=false
05-01 11:49:12.650 15587 15647 W BpBinder: Slow Binder: BpBinder transact took 950 ms, interface=com.android.vending.billing.IInAppBillingService, code=9 oneway=false
--------- beginning of main
05-01 11:49:12.766 15587 15587 W MirrorManager: this model don't Support
05-01 11:49:12.874 15587 15587 W BottomSheetBehavior: Cannot set state: 5
05-01 11:49:12.911 15587 15644 W BpBinder: Slow Binder: BpBinder transact took 2108 ms, interface=com.android.vending.billing.IInAppBillingService, code=9 oneway=false
05-01 11:49:12.912 15587 15645 W BpBinder: Slow Binder: BpBinder transact took 933 ms, interface=com.android.vending.billing.IInAppBillingService, code=9 oneway=false
05-01 11:49:12.927 15587 15587 W libc    : Access denied finding property "[ro.vendor.oplus.market.name](http://ro.vendor.oplus.market.name/)"
05-01 11:49:12.927 15587 15587 W libc    : Access denied finding property "[ro.vendor.oplus.market.name](http://ro.vendor.oplus.market.name/)"
05-01 11:49:12.927 15587 15587 W libc    : Access denied finding property "[ro.vendor.oplus.market.name](http://ro.vendor.oplus.market.name/)"
05-01 11:49:12.965 15587 15587 W ziparchive: Unable to open '/data/app/~~wvs4Q3uKMFsK-ZvHP3HqiQ==/com.google.android.trichromelibrary_561513534-x6zZCsoY7Z-lhoY9-0Pjxw==/[base.dm](http://base.dm/)': No such file or directory
05-01 11:49:12.965 15587 15587 W ziparchive: Unable to open '/data/app/~~wvs4Q3uKMFsK-ZvHP3HqiQ==/com.google.android.trichromelibrary_561513534-x6zZCsoY7Z-lhoY9-0Pjxw==/[base.dm](http://base.dm/)': No such file or directory
05-01 11:49:12.965 15587 15587 W .tagmo.eightbit: Entry not found
05-01 11:49:13.087 15587 15699 W cr_media: BLUETOOTH_CONNECT permission is missing.
05-01 11:49:13.088 15587 15699 W cr_media: registerBluetoothIntentsIfNeeded: Requires BLUETOOTH permission
05-01 11:49:13.152 15587 15706 W AudioCapabilities: Unsupported mime audio/alac
05-01 11:49:13.152 15587 15706 W AudioCapabilities: Unsupported mime audio/x-ape
05-01 11:49:13.152 15587 15706 W AudioCapabilities: Unsupported mime audio/evrc
05-01 11:49:13.152 15587 15706 W AudioCapabilities: Unsupported mime audio/qcelp
05-01 11:49:13.152 15587 15706 W AudioCapabilities: Unsupported mime audio/dsd
05-01 11:49:13.152 15587 15706 W AudioCapabilities: Unsupported mime audio/amr-wb-plus
05-01 11:49:13.152 15587 15706 W AudioCapabilities: Unsupported mime audio/x-ms-wma
05-01 11:49:13.152 15587 15706 W AudioCapabilities: Unsupported mime audio/evrc
05-01 11:49:13.152 15587 15706 W AudioCapabilities: Unsupported mime audio/qcelp
05-01 11:49:13.152 15587 15706 W AudioCapabilities: Unsupported mime audio/alac
05-01 11:49:13.152 15587 15706 W AudioCapabilities: Unsupported mime audio/x-ape
05-01 11:49:13.155 15587 15706 W VideoCapabilities: Unsupported mime image/vnd.android.heic
05-01 11:49:13.155 15587 15706 W VideoCapabilities: Unsupported mime image/vnd.android.heic
05-01 11:49:13.155 15587 15706 W VideoCapabilities: Unsupported mime video/dolby-vision
05-01 11:49:13.155 15587 15706 W VideoCapabilities: Unsupported mime video/dolby-vision
05-01 11:49:13.182 15587 15648 W BpBinder: Slow Binder: BpBinder transact took 2373 ms, interface=com.android.vending.billing.IInAppBillingService, code=9 oneway=false
05-01 11:49:13.334 15587 15646 W BpBinder: Slow Binder: BpBinder transact took 1227 ms, interface=com.android.vending.billing.IInAppBillingService, code=9 oneway=false
05-01 11:49:17.535 15587 15616 W Parcel  : Expecting binder but got null!
05-01 11:49:18.824 15587 15616 W Parcel  : Expecting binder but got null!
05-01 11:49:22.329 15587 15587 W View    : requestLayout() improperly called by androidx.appcompat.widget.AppCompatTextView{63e576f V.ED..... ......ID 189,38-374,109} during layout: running second layout pass
[... cut ...]
05-01 11:49:49.362 15587 15587 W View    : requestLayout() improperly called by android.widget.LinearLayout{62f1312 V.E..V... ........ 0,62-827,98 #7f090116 app:id/group1} during second layout pass: posting in next frame
05-01 11:49:49.362 15587 15587 W View    : requestLayout() improperly called by android.widget.LinearLayout{bc6cae3 V.E..V... ........ 0,98-827,134 #7f090117 app:id/group2} during second layout pass: posting in next frame
05-01 11:49:49.824 15587 15616 W Parcel  : Expecting binder but got null!
```

AbandonedCart, 2023-05-01:
Fixed by 1a47e2acabdb3777108c23cb5c3eed281653df57

closed 2023-05-01 by AbandonedCart