HiddenRamblings/TagMo #673: Crash on wearos

null4n, 2023-05-01:
version #4e86c5d
logcat here
```
 11621 11621 E AndroidRuntime: FATAL EXCEPTION: main
 11621 11621 E AndroidRuntime: Process: com.hiddenramblings.tagmo.eightbit, PID: 11621
 11621 11621 E AndroidRuntime: java.lang.RuntimeException: Unable to instantiate application com.hiddenramblings.tagmo.TagMo: java.lang.NullPointerException: Attempt to invoke virtual method 'android.content.res.Resources android.content.Context.getResources()' on a null object reference
 11621 11621 E AndroidRuntime: 	at android.app.LoadedApk.makeApplication(LoadedApk.java:1256)
 11621 11621 E AndroidRuntime: 	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6714)
 11621 11621 E AndroidRuntime: 	at android.app.ActivityThread.access$1300(ActivityThread.java:241)
 11621 11621 E AndroidRuntime: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1920)
 11621 11621 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:106)
 11621 11621 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:246)
 11621 11621 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:7690)
 11621 11621 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
 11621 11621 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:593)
 11621 11621 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:995)
 11621 11621 E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'android.content.res.Resources android.content.Context.getResources()' on a null object reference
 11621 11621 E AndroidRuntime: 	at android.content.ContextWrapper.getResources(ContextWrapper.java:97)
 11621 11621 E AndroidRuntime: 	at android.content.ContextWrapper.getResources(ContextWrapper.java:97)
 11621 11621 E AndroidRuntime: 	at com.hiddenramblings.tagmo.eightbit.content.ScaledContext.watch(ScaledContext.kt:62)
 11621 11621 E AndroidRuntime: 	at com.hiddenramblings.tagmo.TagMo.<init>(TagMo.kt:79)
 11621 11621 E AndroidRuntime: 	at java.lang.Class.newInstance(Native Method)
 11621 11621 E AndroidRuntime: 	at android.app.AppComponentFactory.instantiateApplication(AppComponentFactory.java:76)
 11621 11621 E AndroidRuntime: 	at androidx.core.app.CoreComponentFactory.instantiateApplication(CoreComponentFactory.java:52)
 11621 11621 E AndroidRuntime: 	at android.app.Instrumentation.newApplication(Instrumentation.java:1158)
 11621 11621 E AndroidRuntime: 	at android.app.LoadedApk.makeApplication(LoadedApk.java:1248)
 11621 11621 E AndroidRuntime: 	... 9 more
```

AbandonedCart, 2023-05-01:
I have a feeling I broke that trying to clean up the startup for phones. The newest build should restore it.

null4n, 2023-05-02:
is it meaning my watch cant use this?
```
18864 18864 W UnsupportedOperationException: java.lang.UnsupportedOperationException
18864 18864 W UnsupportedOperationException:         at android.nfc.NfcAdapter.enableForegroundDispatch(NfcAdapter.java:1543)
18864 18864 W UnsupportedOperationException:         at com.hiddenramblings.tagmo.nfctech.NfcActivity.listenForTags(NfcActivity.kt:647)
18864 18864 W UnsupportedOperationException:         at com.hiddenramblings.tagmo.nfctech.NfcActivity.startNfcMonitor(NfcActivity.kt:618)
18864 18864 W UnsupportedOperationException:         at com.hiddenramblings.tagmo.nfctech.NfcActivity.onResume(NfcActivity.kt:676)
18864 18864 W UnsupportedOperationException:         at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1456)
18864 18864 W UnsupportedOperationException:         at android.app.Activity.performResume(Activity.java:8173)
18864 18864 W UnsupportedOperationException:         at android.app.ActivityThread.performResumeActivity(ActivityThread.java:4445)
18864 18864 W UnsupportedOperationException:         at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:4493)
18864 18864 W UnsupportedOperationException:         at android.app.servertransaction.ResumeActivityItem.execute(ResumeActivityItem.java:52)
18864 18864 W UnsupportedOperationException:         at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:176)
18864 18864 W UnsupportedOperationException:         at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:97)
18864 18864 W UnsupportedOperationException:         at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2073)
18864 18864 W UnsupportedOperationException:         at android.os.Handler.dispatchMessage(Handler.java:106)
18864 18864 W UnsupportedOperationException:         at android.os.Looper.loop(Looper.java:246)
18864 18864 W UnsupportedOperationException:         at android.app.ActivityThread.main(ActivityThread.java:7690)
18864 18864 W UnsupportedOperationException:         at java.lang.reflect.Method.invoke(Native Method)
18864 18864 W UnsupportedOperationException:         at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:593)
18864 18864 W UnsupportedOperationException:         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:995)
```

AbandonedCart, 2023-05-02:
The watches only recently started including NFC, but it doesn't appear to be accessible for writing amiibo. This is still being investigated.

closed 2023-05-02 by AbandonedCart