HiddenRamblings/TagMo #674: N2 Elite randomizer not working proper

NoHonor97, 2023-05-02:
When using the duplicate with random serial feature with N2Elite it randomized it from the selected bin but all the copies are the same as the 1st randomized bin. For example, if I select write to 11 banks with the randomizer checked. It will randomize the serial from the bin in your dump folder, but 11 copies of the same randomized bin.

Is it possible to randomize one at a time? It my take longer but at least it would work.

--------- beginning of main
05-01 17:28:06.345 14117 14117 W BottomSheetBehavior: Cannot set state: 5
05-01 17:28:09.048 14117 14117 W View    : requestLayout() improperly called by androidx.appcompat.widget.AppCompatTextView{463153f V.ED..... ......ID 202,41-397,117} during layout: running second layout pass
05-01 17:28:09.048 14117 14117 W View    : requestLayout() improperly called by androidx.recyclerview.widget.RecyclerView{f4cbdb9 VFED.V... ......ID 0,247-1080,1941 #7f0900dc app:id/elite_content} during layout: running second layout pass
05-01 17:28:09.113 14117 14117 W View    : requestLayout() improperly called by android.widget.LinearLayout{2a2920a V.E..V... ........ 0,0-1024,215} during second layout pass: posting in next frame
05-01 17:28:09.113 14117 14117 W View    : requestLayout() improperly called by android.widget.LinearLayout{44895e V.E..V... ........ 0,67-786,105 #7f090116 app:id/group1} during second layout pass: posting in next frame
05-01 17:28:09.113 14117 14117 W View    : requestLayout() improperly called by android.widget.LinearLayout{bbd513f V.E..V... ........ 0,105-786,143 #7f090117 app:id/group2} during second layout pass: posting in next frame
05-01 17:28:09.113 14117 14117 W View    : requestLayout() improperly called by android.widget.LinearLayout{9eb5862 V.E..V... ........ 0,0-1024,215} during second layout pass: posting in next frame
05-01 17:28:09.113 14117 14117 W View    : requestLayout() improperly called by android.widget.LinearLayout{416fa0c V.E..V... ........ 0,67-786,105 #7f090116 app:id/group1} during second layout pass: posting in next frame
05-01 17:28:09.113 14117 14117 W View    : requestLayout() improperly called by android.widget.LinearLayout{f404f55 V.E..V... ........ 0,105-786,143 #7f090117 app:id/group2} during second layout pass: posting in next frame
05-01 17:28:09.113 14117 14117 W View    : requestLayout() improperly called by android.widget.LinearLayout{10b91ba V.E..V... ......ID 0,0-1024,215} during second layout pass: posting in next frame
05-01 17:28:09.114 14117 14117 W View    : requestLayout() improperly called by android.widget.LinearLayout{74bdd6a V.E..V... ......ID 0,67-786,67 #7f090116 app:id/group1} during second layout pass: posting in next frame
05-01 17:28:09.114 14117 14117 W View    : requestLayout() improperly called by android.widget.LinearLayout{2bfc25b V.E..V... ......ID 0,67-786,67 #7f090117 app:id/group2} during second layout pass: posting in next frame

AbandonedCart, 2023-05-02:
Select the amiibo
Edit properties, randomize serial, save, write, pick slot, repeat

NoHonor97, 2023-05-02:
thank you!!! this works in the meantime, I appreciate all the time and energy you put into to keeping this app going!

AbandonedCart, 2023-05-02:
Hopefully I can figure out why this refuses to set a new UID to each item. It doesn't line up with the code, so there is probably some silent issue that isn't being reported by the debugger.

JigglesTheFett, 2023-05-14:
Has there been any progress on this? I'm having the same issue and it makes me super sad that I can't just write 30 of the same amiibo with different serials quickly.

Aleksxz, 2023-05-14:
Hello again. I really cant find the method (Select amiibo, Edit properties, randomize serial, ...). I pick a random amiibo on n2 elite, and it only show active/replace/write/erase. Nothing else. Maybe because the app is in another language ? I wished that I could switch to english.

AbandonedCart, 2023-05-14:
Click the 3 dots on the right of those.

Aleksxz, 2023-05-14:
There is no dots for me next to the amiibos. Only at the top right of the app and when I click it, this menu pops up with display options.
https://imgur.com/a/jwnkYWT

Aleksxz, 2023-05-14:
Thanks for the update you just did ! Now it show the dots on the Browser display. But somehow, it removed the "delete" button on N2 Elite display, and still not 3 dots there. Screenshot : https://imgur.com/a/qjNZqS5

Anyway thanks for the constant improvements, I really wish you can add a easier method for randomizing serial in the future :)

AbandonedCart, 2023-05-14:
It's under the button

Aleksxz, 2023-05-14:
Probably is, but I can't go up or down in this menu.

AbandonedCart, 2023-05-16:
This issue depends on #677 because until the N2 page stops lagging, individual features are more or less useless.

JigglesTheFett, 2023-05-17:
This did not resolve the issue. It still writes multiple banks with the same serial number.

AbandonedCart, 2023-05-17:
3.8.1 did, though, and that's the build that closed this issue.

JigglesTheFett, 2023-05-17:
Oh! I can only see the 7be3e90 release, which I guess is the 3.8.0. How can I get the 3.8.1?

AbandonedCart, 2023-05-17:
It will pop up in the app automatically. The build needed to be restarted.

CAYCE-VII, 2023-05-17:
3.8.1 still doesn't seem to randomize a bulk write as far as I can tell.

CAYCE-VII, 2023-05-17:
> It still isn't working for me either. I made a quick video to show my process of trying it. I may be doing something wrong, but here's the video: https://imgur.com/a/v2iQRX0

Looks exactly like what I am doing, also N2 Elite. I would love to just be doing something wrong, I'm sure HiddenRamblings would too 😂

AbandonedCart, 2023-05-17:
It's just TagMo. The organization disbanded in 2021.

M0UL, 2023-05-18:
it is not working for me either

RagnarokSam, 2023-05-19:
Also confirming that the "Duplicate with random serial" option for N2 Elite still assigns the same serial number to each bank when writing as of v3.8.3 build #8ae2514

AbandonedCart, 2023-05-19:
I'm aware of the issue.

CAYCE-VII, 2023-05-19:
That kind of works, but when I duplicate 1 amiibo.bin 200 times, the list only shows ~9 of them with anything over that being duplicates of those ~9. It seems like an issue with the list displaying correctly as there are 200 unique files in the directory after using the bin duplicator.

CAYCE-VII, 2023-05-19:
It also looks like the bin duplicator doesn't randomize serials either from what I can tell. All the duplicates have a different serial than the original, but the duplicates all have the same serial.

AbandonedCart, 2023-05-19:
Interesting.

AbandonedCart, 2023-05-28:
Fixed by 754e020

closed 2023-05-28 by AbandonedCart

random11x, 2023-06-29:
i assume this fix would be live... however i am still having this issue. I enable the randomize on clone option and all amiibo still have the same serial.