HiddenRamblings/TagMo #677: [Issue]: Recent versions of TagMo hang shortly after starting

Garonne, 2023-05-14:
### Issue Description

I just reinstalled TagMo after not needing it for quite some time. After about 2-3 minutes the app hangs for about 60 seconds and then repeatedly hangs again for a couple of seconds about every minute.

Using a Samsung Galaxy S20 FE Smpartphone

### Logcat Output

```shell
No Crashlog available...
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

Garonne, 2023-05-14:
Figured out how to find Logcat and added it to the description.
Also probably found N2Elite to be triggering issue.

Garonne, 2023-05-14:
Actually found two more logcats TagMo automatically copied to clipboard:

May 14th, 5:02 pm:
```shell

TagMo 3.7.9 (GitHub Release) #4c39598
samsung TIRAMISU (13) - 5.00 GB RAM

--------- beginning of main
05-14 16:52:41.194 25962 25991 W Parcel  : Expecting binder but got null!
```

May 14th, 5:04 pm:
```shell

TagMo 3.7.9 (GitHub Release) #4c39598
samsung TIRAMISU (13) - 5.00 GB RAM

--------- beginning of main
05-14 17:02:31.346  1903  1903 W BottomSheetBehavior: Cannot set state: 5
05-14 17:02:31.347  1903  1903 W View    : requestLayout() improperly called by androidx.appcompat.widget.AppCompatTextView{41f96b2 V.ED..... ......ID 216,43-424,124} during layout: running second layout pass
05-14 17:02:31.347  1903  1903 W View    : requestLayout() improperly called by androidx.recyclerview.widget.RecyclerView{2eec8c8 VFED.V... ......ID 0,264-1080,1793 #7f0900db app:id/elite_content} during layout: running second layout pass
05-14 17:02:31.620  1903  3171 W VideoCapabilities: Unsupported mime image/vnd.android.heic
05-14 17:02:31.623  1903  3171 W AudioCapabilities: Unsupported mime audio/x-ape
05-14 17:02:31.624  1903  3171 W AudioCapabilities: Unsupported mime audio/x-ima
05-14 17:02:31.624  1903  3171 W AudioCapabilities: Unsupported mime audio/mpeg-L1
05-14 17:02:31.624  1903  3171 W AudioCapabilities: Unsupported mime audio/mpeg-L2
05-14 17:02:31.624  1903  3171 W VideoCapabilities: Unsupported mime video/mp43
05-14 17:02:31.624  1903  3171 W VideoCapabilities: Unsupported mime video/wvc1
05-14 17:02:31.625  1903  3171 W VideoCapabilities: Unsupported mime video/x-ms-wmv
05-14 17:02:31.625  1903  3171 W AudioCapabilities: Unsupported mime audio/x-ms-wma
05-14 17:02:31.625  1903  3171 W VideoCapabilities: Unsupported mime video/x-ms-wmv7
05-14 17:02:31.625  1903  3171 W VideoCapabilities: Unsupported mime video/x-ms-wmv8
05-14 17:03:17.391  1903  1936 W Parcel  : Expecting binder but got null!
```

Last Logcat (added to the first post) was created shortly after this one.

AbandonedCart, 2023-05-15:
I'm not entirely sure that ANRWatchdog actually has any value. The N2 page being where it starts is the part that was known, but the exact cause has evaded nearly a full rewrite of the code.

AbandonedCart, 2023-05-15:
More specifically, the issue appears to be in the bottom drawer of the N2 page.

Garonne, 2023-05-15:
Any idea when this issue first occured? My very old TagMo 2.7.5 version works fine... It's just unable to identify recent Amiibos (like the new Zelda TotK Amiibo).

AbandonedCart, 2023-05-16:
If you scan an N2, switch to device options, then swipe the numbers, does it start to lag on the 6th swipe?

AbandonedCart, 2023-05-17:
As a temporary solution, if "force software layer" is not already enabled in settings, enable it. The number pickers will stop snapping into place, but it will solve the lag.

closed 2023-05-17 by AbandonedCart

Garonne, 2023-05-22:
"Force software layer" didn't fix it. But I still have TagMo 2.7.5 as backup.

Logcat 1:
```shell
TagMo 3.8.4 (GitHub Release) #ad66a2a
samsung TIRAMISU (13) - 5.00 GB RAM

--------- beginning of main
05-22 15:30:12.351 23692 23692 W ziparchive: Unable to open '/data/app/~~N4Ql6ka240-In84vD9BfBA==/com.google.android.trichromelibrary_567213133-tti68dpuNDPVo5Unk1fLzw==/base.dm': No such file or directory
05-22 15:30:12.351 23692 23692 W ziparchive: Unable to open '/data/app/~~N4Ql6ka240-In84vD9BfBA==/com.google.android.trichromelibrary_567213133-tti68dpuNDPVo5Unk1fLzw==/base.dm': No such file or directory
05-22 15:30:12.351 23692 23692 W .tagmo.eightbit: Entry not found
05-22 15:30:12.804 23692 24148 W cr_media: BLUETOOTH_CONNECT permission is missing.
05-22 15:30:12.805 23692 24148 W cr_media: registerBluetoothIntentsIfNeeded: Requires BLUETOOTH permission
05-22 15:30:13.190 23692 24154 W VideoCapabilities: Unsupported mime image/vnd.android.heic
05-22 15:30:13.197 23692 24154 W AudioCapabilities: Unsupported mime audio/x-ape
05-22 15:30:13.197 23692 24154 W AudioCapabilities: Unsupported mime audio/x-ima
05-22 15:30:13.197 23692 24154 W AudioCapabilities: Unsupported mime audio/mpeg-L1
05-22 15:30:13.197 23692 24154 W AudioCapabilities: Unsupported mime audio/mpeg-L2
05-22 15:30:13.197 23692 24154 W VideoCapabilities: Unsupported mime video/mp43
05-22 15:30:13.197 23692 24154 W VideoCapabilities: Unsupported mime video/wvc1
05-22 15:30:13.197 23692 24154 W VideoCapabilities: Unsupported mime video/x-ms-wmv
05-22 15:30:13.198 23692 24154 W AudioCapabilities: Unsupported mime audio/x-ms-wma
05-22 15:30:13.198 23692 24154 W VideoCapabilities: Unsupported mime video/x-ms-wmv7
05-22 15:30:13.198 23692 24154 W VideoCapabilities: Unsupported mime video/x-ms-wmv8
05-22 15:30:42.963 23692 23728 W Parcel  : Expecting binder but got null!
```

Logcat 2:
```shell
TagMo 3.8.4 (GitHub Release) #ad66a2a
samsung TIRAMISU (13) - 5.00 GB RAM

--------- beginning of main
05-22 15:35:19.374 28586 28707 E chromium: [ERROR:simple_file_enumerator.cc(21)] opendir /data/user/0/com.hiddenramblings.tagmo.eightbit/cache/WebView/Default/HTTP Cache/Code Cache/wasm: No such file or directory (2)
05-22 15:35:19.374 28586 28707 E chromium: [ERROR:simple_index_file.cc(614)] Could not reconstruct index from disk
05-22 15:35:19.415 28586 28726 W cr_media: BLUETOOTH_CONNECT permission is missing.
05-22 15:35:19.415 28586 28726 W cr_media: registerBluetoothIntentsIfNeeded: Requires BLUETOOTH permission
05-22 15:35:19.593 28586 28733 W VideoCapabilities: Unsupported mime image/vnd.android.heic
05-22 15:35:19.595 28586 28733 W AudioCapabilities: Unsupported mime audio/x-ape
05-22 15:35:19.595 28586 28733 W AudioCapabilities: Unsupported mime audio/x-ima
05-22 15:35:19.596 28586 28733 W AudioCapabilities: Unsupported mime audio/mpeg-L1
05-22 15:35:19.596 28586 28733 W AudioCapabilities: Unsupported mime audio/mpeg-L2
05-22 15:35:19.596 28586 28733 W VideoCapabilities: Unsupported mime video/mp43
05-22 15:35:19.596 28586 28733 W VideoCapabilities: Unsupported mime video/wvc1
05-22 15:35:19.596 28586 28733 W VideoCapabilities: Unsupported mime video/x-ms-wmv
05-22 15:35:19.596 28586 28733 W AudioCapabilities: Unsupported mime audio/x-ms-wma
05-22 15:35:19.596 28586 28733 W VideoCapabilities: Unsupported mime video/x-ms-wmv7
05-22 15:35:19.596 28586 28733 W VideoCapabilities: Unsupported mime video/x-ms-wmv8
05-22 15:36:12.261 28586 28606 W Parcel  : Expecting binder but got null!
```

AbandonedCart, 2023-05-22:
2.7.5 isn't really a solution, given the number of changes. It appears your issue is with bluetooth. That's interesting.

Garonne, 2023-05-30:
Did you remove Flask support between 3.8.4 and 3.9.1? Anyways, 3.9.1 seems to work flawlessly with N2Elite again.

AbandonedCart, 2023-05-30:
It's labeled "Bluetooth" now because there are multiple devices.

Garonne, 2023-05-30:
Yep. Found it. It's labeled "Bluetooth-Browser" in my edition. Well. Issue fixed. Nothing more from my side.

AbandonedCart, 2023-05-30:
The menu button is "Bluetooth Browser" and the page appears as "Bluetooth" but you can get there by either clicking the menu or swiping horizontally. It no longer requires being enabled, since enabling and disabling it had no impact on anything other than the previous tab being visible.