HiddenRamblings/TagMo #683: Guide page shows Webpage not available and trying to switch to the browser cases the app to ctd as well as everytime I try to open the app after that forceing me to reinstall..

Zippomoon, 2023-05-18:
Phone model: HTC Evo 4G LTE
Android version: 4.1.1

The Github version is even worse the the Google Play version as it just CTDs with out even showing the menu.

Bought this phone back in 2018 for the express purpose of using it for this app, hope this is something a update can fix as I'd rather not have to buy a new phone.

Zippomoon, 2023-05-18:
TagMo 3.8.1 (Play Release) #7be3e903
HTC JELLY_BEAN (4.1.1) - 777.00 MB RAMjava.lang.NoSuchMethodError: android.widget.FrameLayout$LayoutParams.setMarginEnd
	at me.zhanghai.android.fastscroll.PopupStyles$$ExternalSyntheticApiModelOutline0.m(R8$$SyntheticClass:0)
	at me.zhanghai.android.fastscroll.PopupStyles.lambda$static$0(PopupStyles.java:41)
	at me.zhanghai.android.fastscroll.PopupStyles.$r8$lambda$GsThvF5qiWH0MR_ju6p2DG-w94c(PopupStyles.java:0)
	at me.zhanghai.android.fastscroll.PopupStyles$$ExternalSyntheticLambda2.accept(R8$$SyntheticClass:0)
	at me.zhanghai.android.fastscroll.FastScroller.<init>(FastScroller.java:110)
	at me.zhanghai.android.fastscroll.FastScrollerBuilder.build(FastScrollerBuilder.java:151)
	at com.hiddenramblings.tagmo.browser.fragment.BrowserFragment.onViewCreated(BrowserFragment.kt:143)
	at androidx.fragment.app.Fragment.performViewCreated(Fragment.java:3137)
	at androidx.fragment.app.FragmentStateManager.createView(FragmentStateManager.java:552)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:261)
	at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:1901)
	at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:1819)
	at androidx.fragment.app.FragmentManager.execSingleAction(FragmentManager.java:1731)
	at androidx.fragment.app.BackStackRecord.commitNow(BackStackRecord.java:317)
	at androidx.viewpager2.adapter.FragmentStateAdapter.placeFragmentInViewHolder(FragmentStateAdapter.java:341)
	at androidx.viewpager2.adapter.FragmentStateAdapter.onViewAttachedToWindow(FragmentStateAdapter.java:276)
	at androidx.viewpager2.adapter.FragmentStateAdapter.onViewAttachedToWindow(FragmentStateAdapter.java:67)
	at androidx.recyclerview.widget.RecyclerView.dispatchChildAttached(RecyclerView.java:8291)
	at androidx.recyclerview.widget.RecyclerView$5.addView(RecyclerView.java:924)
	at androidx.recyclerview.widget.ChildHelper.addView(ChildHelper.java:107)
	at androidx.recyclerview.widget.RecyclerView$LayoutManager.addViewInt(RecyclerView.java:9344)
	at androidx.recyclerview.widget.RecyclerView$LayoutManager.addView(RecyclerView.java:9302)
	at androidx.recyclerview.widget.RecyclerView$LayoutManager.addView(RecyclerView.java:9289)
	at androidx.recyclerview.widget.LinearLayoutManager.layoutChunk(LinearLayoutManager.java:1676)
	at androidx.recyclerview.widget.LinearLayoutManager.fill(LinearLayoutManager.java:1622)
	at androidx.recyclerview.widget.LinearLayoutManager.onLayoutChildren(LinearLayoutManager.java:687)
	at androidx.recyclerview.widget.RecyclerView.dispatchLayoutStep2(RecyclerView.java:4604)
	at androidx.recyclerview.widget.RecyclerView.dispatchLayout(RecyclerView.java:4307)
	at androidx.recyclerview.widget.RecyclerView.onLayout(RecyclerView.java:4873)
	at android.view.View.layout(View.java:13836)
	at android.view.ViewGroup.layout(ViewGroup.java:4380)
	at androidx.viewpager2.widget.ViewPager2.onLayout(ViewPager2.java:527)
	at android.view.View.layout(View.java:13836)
	at android.view.ViewGroup.layout(ViewGroup.java:4380)
	at androidx.coordinatorlayout.widget.CoordinatorLayout.layoutChild(CoordinatorLayout.java:1213)
	at androidx.coordinatorlayout.widget.CoordinatorLayout.onLayoutChild(CoordinatorLayout.java:899)
	at androidx.coordinatorlayout.widget.CoordinatorLayout.onLayout(CoordinatorLayout.java:919)
	at android.view.View.layout(View.java:13836)
	at android.view.ViewGroup.layout(ViewGroup.java:4380)
	at android.widget.LinearLayout.setChildFrame(LinearLayout.java:1649)
	at android.widget.LinearLayout.layoutVertical(LinearLayout.java:1507)
	at android.widget.LinearLayout.onLayout(LinearLayout.java:1420)
	at android.view.View.layout(View.java:13836)
	at android.view.ViewGroup.layout(ViewGroup.java:4380)
	at androidx.drawerlayout.widget.DrawerLayout.onLayout(DrawerLayout.java:1263)
	at android.view.View.layout(View.java:13836)
	at android.view.ViewGroup.layout(ViewGroup.java:4380)
	at android.widget.FrameLayout.onLayout(FrameLayout.java:448)
	at android.view.View.layout(View.java:13836)
	at android.view.ViewGroup.layout(ViewGroup.java:4380)
	at androidx.appcompat.widget.ActionBarOverlayLayout.onLayout(ActionBarOverlayLayout.java:536)
	at android.view.View.layout(View.java:13836)
	at android.view.ViewGroup.layout(ViewGroup.java:4380)
	at android.widget.FrameLayout.onLayout(FrameLayout.java:448)
	at android.view.View.layout(View.java:13836)
	at android.view.ViewGroup.layout(ViewGroup.java:4380)
	at android.widget.LinearLayout.setChildFrame(LinearLayout.java:1649)
	at android.widget.LinearLayout.layoutVertical(LinearLayout.java:1507)
	at android.widget.LinearLayout.onLayout(LinearLayout.java:1420)
	at android.view.View.layout(View.java:13836)
	at android.view.ViewGroup.layout(ViewGroup.java:4380)
	at android.widget.FrameLayout.onLayout(FrameLayout.java:448)
	at android.view.View.layout(View.java:13836)
	at android.view.ViewGroup.layout(ViewGroup.java:4380)
	at android.view.ViewRootImpl.performLayout(ViewRootImpl.java:2112)
	at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:1913)
	at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1154)
	at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:4667)
	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:734)
	at android.view.Choreographer.doCallbacks(Choreographer.java:560)
	at android.view.Choreographer.doFrame(Choreographer.java:529)
[... cut ...]
	at android.os.Handler.handleCallback(Handler.java:615)
	at android.os.Handler.dispatchMessage(Handler.java:92)
	at android.os.Looper.loop(Looper.java:155)
	at android.app.ActivityThread.main(ActivityThread.java:5454)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:511)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1029)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:796)
	at dalvik.system.NativeStart.main(Native Method)

AbandonedCart, 2023-05-18:
One of the libraries was incompatible because your phone is 11 years old. While an attempt it made to continue supporting older Android versions, they are not regularly tested. Issues are resolved when they are reported.

closed 2023-05-18 by AbandonedCart