HiddenRamblings/TagMo #690: freezes after selecting folder

irvingbenja, 2023-05-22:
### Issue Description

After updating to The newest version this afternoon, tagmo has completely frozen and crashed. Initially upon updating it would show the window and then completely freeze up. After your uninstalling it, restarting my phone, and reinstalling, I can now bring up the file browser for the Bin files. As soon as I click use this folder it completely freezes again, eventually crashing and providing a log.

My phone is a Verizon Galaxy s21 unrooted

### Logcat Output

```shell
Will submit via discord, I'm getting a character limit error when submitting through this form.
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

bulldog93, 2023-05-23:
Same issue here. Oneplus 7 Pro

AbandonedCart, 2023-05-23:
Fixed by 2755c65616f02d240645c48b1fdf9b5318d9c59d (Just waiting on the rate limit timer)

closed 2023-05-23 by AbandonedCart