HiddenRamblings/TagMo #694: [Issue]: amiibo creation stopped working

Billbobbybill, 2023-05-23:
### Issue Description

After last update, writing an amiibo does not work. (Happened after today's update).

### Logcat Output

```shell
TagMo 3.8.6 (Play Release) #a98dc9c3
samsung TIRAMISU (13) - 6.00 GB RAM

--------- beginning of main
05-22 23:17:56.791 27696 27696 E BufferQueueProducer: Unable to open libpenguin.so: dlopen failed: library "libpenguin.so" not found.
05-22 23:17:56.800 27696 27696 W View    : requestLayout() improperly called by android.widget.FrameLayout{40b8252 V.E..V... ......ID 0,0-1080,1947 #2} during layout: running second layout pass
05-22 23:17:56.800 27696 27696 W View    : requestLayout() improperly called by androidx.recyclerview.widget.RecyclerView{2b50f50 VFED.V... ......ID 0,1824-1080,1824 #7f090108 app:id/foomiibo_list} during layout: running second layout pass
05-22 23:17:56.800 27696 27696 W View    : requestLayout() improperly called by androidx.recyclerview.widget.RecyclerView{22a48e4 VFED.V... ......ID 0,0-1080,1812 #7f090080 app:id/browser_content} during layout: running second layout pass
05-22 23:17:58.705 27696 27710 W JavaBinder: BinderProxy is being destroyed but the application did not call unlinkToDeath to unlink all of its death recipients beforehand.  Releasing leaked death recipient: com.google.android.play.core.internal.zzai
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2023-05-23:
Fixed by 51ef32542a3c419555cd3b0055425d256ffe38ee

Billbobbybill, 2023-05-24:
Update: when scanning, the scanning process gets on step further than it originally was with the issue, but it could have just been going fast so I couldn't tell.

AbandonedCart, 2023-05-24:
Working without any log of the actual issue makes it a bit harder to identify.

Billbobbybill, 2023-05-24:
What can I do to help?

AbandonedCart, 2023-05-24:
Try to do whatever is failing, then immediately get the log. That will make sure it's included.

Billbobbybill, 2023-05-24:
Let me know if I can do anything else to help out

closed 2023-05-25 by AbandonedCart