HiddenRamblings/TagMo #696: Write tag option not becoming avaliable

DeerlyNostalgic, 2023-05-24:
### Issue Description

The write tag option is staying blacked out even though I have the amiibo zip ready to go on scream. It's also asking for a new secret key and saying all the ones I've put in are not recognized. 

### Logcat Output

```shell
The write tag option is staying blacked out even though I have the amiibo zip ready to go on scream. It's also asking for a new secret key and saying all the ones I've put in are not recognized.
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2023-05-24:
The app doesn't support zip files and issues without logs cannot be investigated further.

closed 2023-05-24 by AbandonedCart

DeerlyNostalgic, 2023-05-24:
Wait but is there a way to fix the issue?