HiddenRamblings/TagMo #697: [Issue]: log that goes with the other issue

Billbobbybill, 2023-05-24:
### Issue Description

Here is the log

### Logcat Output

```shell
TagMo 3.8.8 (Play Release) #7131ba6a
samsung TIRAMISU (13) - 6.00 GB RAM

--------- beginning of main
05-23 21:39:12.962  8197  8437 W TagLostException: android.nfc.TagLostException: Tag was lost.
05-23 21:39:12.962  8197  8437 W TagLostException: 	at android.nfc.TransceiveResult.getResponseOrThrow(TransceiveResult.java:48)
05-23 21:39:12.962  8197  8437 W TagLostException: 	at android.nfc.tech.BasicTagTechnology.transceive(BasicTagTechnology.java:154)
05-23 21:39:12.962  8197  8437 W TagLostException: 	at android.nfc.tech.MifareUltralight.transceive(MifareUltralight.java:215)
05-23 21:39:12.962  8197  8437 W TagLostException: 	at com.hiddenramblings.tagmo.nfctech.NTAG215.transceive(NTAG215.kt:66)
05-23 21:39:12.962  8197  8437 W TagLostException: 	at com.hiddenramblings.tagmo.nfctech.NTAG215.readSignature(NTAG215.kt:122)
05-23 21:39:12.962  8197  8437 W TagLostException: 	at com.hiddenramblings.tagmo.nfctech.TagArray.isElite(TagArray.kt:83)
05-23 21:39:12.962  8197  8437 W TagLostException: 	at com.hiddenramblings.tagmo.nfctech.NfcActivity.onTagDiscovered(NfcActivity.kt:241)
05-23 21:39:12.962  8197  8437 W TagLostException: 	at com.hiddenramblings.tagmo.nfctech.NfcActivity.onNewIntent$lambda$41(NfcActivity.kt:678)
05-23 21:39:12.962  8197  8437 W TagLostException: 	at com.hiddenramblings.tagmo.nfctech.NfcActivity.$r8$lambda$GuAJJzT4ercv9FKRB4lwPVLYAw0(NfcActivity.kt:0)
05-23 21:39:12.962  8197  8437 W TagLostException: 	at com.hiddenramblings.tagmo.nfctech.NfcActivity$$ExternalSyntheticLambda5.run(R8$$SyntheticClass:0)
05-23 21:39:12.962  8197  8437 W TagLostException: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
05-23 21:39:12.962  8197  8437 W TagLostException: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
05-23 21:39:12.962  8197  8437 W TagLostException: 	at java.lang.Thread.run(Thread.java:1012)
05-23 21:39:13.612  8197  8437 W TagLostException: android.nfc.TagLostException: Tag was lost.
05-23 21:39:13.612  8197  8437 W TagLostException: 	at android.nfc.TransceiveResult.getResponseOrThrow(TransceiveResult.java:48)
05-23 21:39:13.612  8197  8437 W TagLostException: 	at android.nfc.tech.BasicTagTechnology.transceive(BasicTagTechnology.java:154)
05-23 21:39:13.612  8197  8437 W TagLostException: 	at android.nfc.tech.MifareUltralight.readPages(MifareUltralight.java:164)
05-23 21:39:13.612  8197  8437 W TagLostException: 	at com.hiddenramblings.tagmo.nfctech.NTAG215.readPages(NTAG215.kt:82)
05-23 21:39:13.612  8197  8437 W TagLostException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.readFromTag(TagReader.kt:68)
05-23 21:39:13.612  8197  8437 W TagLostException: 	at com.hiddenramblings.tagmo.nfctech.NfcActivity.onTagDiscovered(NfcActivity.kt:303)
05-23 21:39:13.612  8197  8437 W TagLostException: 	at com.hiddenramblings.tagmo.nfctech.NfcActivity.onNewIntent$lambda$41(NfcActivity.kt:678)
05-23 21:39:13.612  8197  8437 W TagLostException: 	at com.hiddenramblings.tagmo.nfctech.NfcActivity.$r8$lambda$GuAJJzT4ercv9FKRB4lwPVLYAw0(NfcActivity.kt:0)
05-23 21:39:13.612  8197  8437 W TagLostException: 	at com.hiddenramblings.tagmo.nfctech.NfcActivity$$ExternalSyntheticLambda5.run(R8$$SyntheticClass:0)
05-23 21:39:13.612  8197  8437 W TagLostException: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
05-23 21:39:13.612  8197  8437 W TagLostException: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
05-23 21:39:13.612  8197  8437 W TagLostException: 	at java.lang.Thread.run(Thread.java:1012)
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2023-05-24:
According to the error, you're pulling the N2 away from the phone too fast.

AbandonedCart, 2023-05-24:
Merged with #694

closed 2023-05-24 by AbandonedCart