HiddenRamblings/TagMo #699: [Feature]: Consider adding a converter for FlipperZero (BIN to NFC) with random ID option.

teknomedic, 2023-06-01:
### Feature Explanation

Having a built in Amiibo data file converter would be a nice addition such as on the fly converting (based on settings set by user) for a device like the FlipperZero.

Been looking for an All-in-one solution for taking a retail Amiibo BIN data and then cloning it several times with random IDs (making say 20x Zelda bins with different IDs) for use on non-modded consoles.  This would help to get special items in games like Breath of the Wild or Tears of the Kingdom that only allow an Amiibo to be used once in 24 hours. Currently one needs to save, try an Amiibo and then re-load previous save to try again if it failed.

I guess I visualize the process as something like... "load Zelda.bin, set output to FlipperZero format (Zelda.nfc), randomize IDs set to yes, quantity = 20, Start"... then the output being "zelda01.nfc, zelda02.nfc, zelda03.nfc, etc".

### Usage Confirmation

- [X] I confirm this is a request for a new feature, not an issue

AbandonedCart, 2023-06-01:
As it stands, there are no plans to add support for the Flipper Zero, since it uses non-standard formats and there is no way to test due to availability.

AbandonedCart, 2023-09-06:
This may or may not be added in a future update, depending on whether or not it will create the possibility to exploit other features, such as Foomiibo.

closed 2023-09-06 by AbandonedCart

AbandonedCart, 2023-09-07:
Since it was possible to add the signature that identifies the files to the Flipper files, there is now a button to export. It doesn't do bulk, but it at least allows exporting them. The process to accomplish the task mentioned in the request would be to use the bin duplicator and then export the files. I will investigate the possibility of doing a bulk export.

teknomedic, 2023-09-08:
Wow, thank you for the updates.  Earlier this year I did bend the ear of another programer and they added a few of the requested features to their program.  Here's a link if you'd like to reference their project or discuss with them ideas related to this since the projects are similar....
https://github.com/DarkIrata/AmiiboSN-Changer

AbandonedCart, 2023-09-08:
I've seen that project, but changing serial numbers is only a small portion of TagMo. The issue isn't the features, which are extremely simple, but finding room for them in an already crowded UI.

AbandonedCart, 2023-09-08:
The bulk option will be under the "Bin Randomizer"