HiddenRamblings/TagMo #703: [Issue]: NFC Scan button missing after closing/reopening app

infiCathi, 2023-06-11:
### Issue Description

No Scanner NFC button in bottom right corner of screen once I initially close the app. Have to reinstall to get it to come back. 

Also couldn't search banks to assign, but that was probably a me issue. Did a combo of the n2 elite app + this one (this could read what was in the banks, n2 would show blank, but I could assign). 

): Hope this info helps at all?? 

### Logcat Output

```shell
TagMo 3.9.4 (Play Release) #b259982b
samsung TIRAMISU (13) - 10.00 GB RAM

--------- beginning of main
06-11 13:26:53.564   841   850 W .tagmo.eightbit: Reducing the number of considered missed Gc histogram windows from 117 to 100
06-11 13:27:12.368   841   841 W ziparchive: Unable to open '/data/app/~~67355qvpDfgsx5UXQzW3fA==/com.google.android.trichromelibrary_567216233-eIv98WmyTgVh0tBknbHUiw==/base.dm': No such file or directory
06-11 13:27:12.368   841   841 W ziparchive: Unable to open '/data/app/~~67355qvpDfgsx5UXQzW3fA==/com.google.android.trichromelibrary_567216233-eIv98WmyTgVh0tBknbHUiw==/base.dm': No such file or directory
06-11 13:27:12.368   841   841 W .tagmo.eightbit: Entry not found
06-11 13:27:12.443   841   851 W JavaBinder: BinderProxy is being destroyed but the application did not call unlinkToDeath to unlink all of its death recipients beforehand.  Releasing leaked death recipient: com.google.android.play.core.internal.zzai
06-11 13:27:12.455   841  3050 W cr_media: BLUETOOTH_CONNECT permission is missing.
06-11 13:27:12.455   841  3050 W cr_media: registerBluetoothIntentsIfNeeded: Requires BLUETOOTH permission
06-11 13:27:12.512   841  3086 W AudioCapabilities: Unsupported mime audio/x-ape
06-11 13:27:12.512   841  3086 W AudioCapabilities: Unsupported mime audio/evrc
06-11 13:27:12.512   841  3086 W AudioCapabilities: Unsupported mime audio/evrc
06-11 13:27:12.513   841  3086 W AudioCapabilities: Unsupported mime audio/x-ima
06-11 13:27:12.513   841  3086 W AudioCapabilities: Unsupported mime audio/mpeg-L1
06-11 13:27:12.513   841  3086 W AudioCapabilities: Unsupported mime audio/mpeg-L2
06-11 13:27:12.513   841  3086 W AudioCapabilities: Unsupported mime audio/qcelp
06-11 13:27:12.513   841  3086 W AudioCapabilities: Unsupported mime audio/qcelp
06-11 13:27:12.513   841  3086 W VideoCapabilities: Unsupported mime video/wvc1
06-11 13:27:12.513   841  3086 W VideoCapabilities: Unsupported mime video/x-ms-wmv
06-11 13:27:12.516   841  3086 W VideoCapabilities: Unsupported mime image/vnd.android.heic
06-11 13:27:12.516   841  3086 W VideoCapabilities: Unsupported mime image/vnd.android.heic
06-11 13:27:12.516   841  3086 W VideoCapabilities: Unrecognized profile/level 32768/256 for video/mp4v-es
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2023-06-11:
Are you switching back and forth between portrait and landscape? I'm working to solve that issue once, but there is a holdup with the build machine being out of commission.

infiCathi, 2023-06-12:
> working to solve that issue, but there is a holdup with the build machine

No, just portrait. It's literally just whenever I close the app, if I just
keep it open the button stays there for hours. Not sure if there's a button
that can be added in for it, or if it's just N2 elite feature then toggling
it in and out if you turn that on and off might help? :o 😅

AbandonedCart, 2023-06-12:
I have a feeling it's just the refresh being called at the wrong time.

infiCathi, 2023-06-12:
Clearing data seems to have worked, it stays on reload now. Don't have n2 elite tag on me but will test in a few hours to see if clearing just cache might work.  s23 ultra if that helps any

AbandonedCart, 2023-06-12:
I'm not sure what caused it then, but I've taken some steps to minimize the chances it will happen again. They will reach Google Play after a little further testing.

AbandonedCart, 2023-06-25:
After some thorough testing, this appears to have been resolved.

closed 2023-06-25 by AbandonedCart