HiddenRamblings/TagMo #705: [Issue]: amiibos not being displayed

TheTaterLord, 2023-06-18:
### Issue Description

I have over 200+ amiibo bin files in my collection and tagmo apparently can only see my link TOTK one I've tried reinstalling the app I've tried the newest GitHub build as well both have the same issue

### Logcat Output

```shell
Not sure what this is or how to obtain it
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2023-06-19:
Did you load your keys? Is there a filter active?

AbandonedCart, 2023-06-25:
This appears to have been resolved for the specific instance and steps were taken to prevent it for new / returning users, at the possible risk of displaying duplicates. Two of one is better than none, though.

closed 2023-06-25 by AbandonedCart