HiddenRamblings/TagMo #708: [Issue]: reducing the bank count on an n2elite is buggy

random11x, 2023-06-29:
### Issue Description

When I reduce the bank count the App seems to successfully write the change but the UI still shows the extra banks as populated. Even rescanning the tag does not correct the issue. You have to force quit the app and then rescan it to see the change.

### Logcat Output

```shell
Im submitting this on pc I cant paste any logcat output
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2023-06-29:
The log is automatically copied to the clipboard, as described in the https://github.com/HiddenRamblings/TagMo/wiki#reporting-issues page

random11x, 2023-06-29:
I have updated the post to have the logcat output. I reproduced the problem and then pasted the output.

AbandonedCart, 2023-06-29:
No need to clarify. That's the purpose.

random11x, 2023-07-03:
I downloaded "4.0.0". Unfortunately I still get the same behaviour for all 3 issues.

AbandonedCart, 2023-07-04:
Sounds like a hardware issue then.

CurtisDS, 2023-07-04:
With my phone or the n2elite? I have access to another phone I could test. As for the n2Elite, I can write data to it fine using the standard steps:

Scan the n2elite, choose a slot, replace it with a new bin. etc.

I can also edit each bin one by one editing the properties and manually choosing to create a random serial.

So if that works, I'm not sure there would be an issue with the n2elite itself.

I'll go test the other phone and let you know.

random11x, 2023-07-04:
With my phone or the n2elite? I have access to another phone I could test. As for the n2Elite, I can write data to it fine using the standard steps:

Scan the n2elite, choose a slot, replace it with a new bin. etc.

I can also edit each bin one by one editing the properties and manually choosing to create a random serial.

So if that works, I'm not sure there would be an issue with the n2elite itself.

I'll go test the other phone and let you know.

random11x, 2023-07-04:
Same exact thing with the other phone.

This was the LogCat info:

```
TagMo 4.0.0 (GitHub Release) #e5906df
Google R (11) - 3.00 GB RAM

--------- beginning of main
07-03 21:49:36.146 10001 10001 E .tagmo.eightbi: Not starting debugger since process cannot load the jdwp agent.
07-03 21:49:36.704 10001 10021 W AdrenoUtils: <ReadGpuID_from_sysfs:197>: Failed to open /sys/class/kgsl/kgsl-3d0/gpu_model
07-03 21:49:36.704 10001 10021 W AdrenoUtils: <ReadGpuID:221>: Failed to read chip ID from gpu_model. Fallback to use the GSL path
07-03 21:49:36.744 10001 10001 W View    : requestLayout() improperly called by android.widget.FrameLayout{7dff23e V.E..V... ......ID 0,0-1440,2418 #2} during layout: running second layout pass
07-03 21:49:36.744 10001 10001 W View    : requestLayout() improperly called by androidx.recyclerview.widget.RecyclerView{1d389f VFED.V... ......ID 0,2428-1440,2428 #7f09010c app:id/foomiibo_list} during layout: running second layout pass
07-03 21:49:36.744 10001 10001 W View    : requestLayout() improperly called by androidx.recyclerview.widget.RecyclerView{dabbdec VFED.V... ......ID 0,0-1440,2414 #7f090081 app:id/browser_content} during layout: running second layout pass
07-03 21:49:36.758 10001 10021 W Gralloc3: mapper 3.x is not supported
```

AbandonedCart, 2023-07-17:
Seems the issue is the list itself not adjusting the number of rows. It may be related to Android trying to phase out notifydatasetchanged in favor of added, removed, and changed. It wouldn't be the first time they forced everyone to move to a new design and didn't test, complete it, etc.

AbandonedCart, 2023-07-23:
Are you able to increase the bank count without box issues?

CurtisDS, 2023-07-24:
> Are you able to increase the bank count without box issues?
On version #da8ff77 it still doesnt update the number of banks shown until you force quit the app and restart it.

random11x, 2023-07-24:
> Are you able to increase the bank count without box issues?

On version #da8ff77 it still doesn't update the number of banks shown until you force quit the app and restart it.

AbandonedCart, 2023-07-24:
So increasing doesn't update, either?

random11x, 2023-07-24:
increasing works. But it always has worked. The bug only happens when reducing the bank count.

AbandonedCart, 2023-08-02:
There weren't any changes for that in the latest build.

random11x, 2023-08-02:
oh...well this is the first Ive had it fully crash when shrinking the bank count. So I figured I would give an update

AbandonedCart, 2023-08-04:
Somehow the reduced list became just the items that were being removed, which meant the number of items was actually how many need to be taken away from the list instead of how many should be left.

TL;DR the values ended up backwards, but it should be fixed now.

commit 22ae70ef (2023-08-10): Fix bulk writing random serial to N2 Elite

Closes #708
closed 2023-08-10 by AbandonedCart