HiddenRamblings/TagMo #709: [Feature]: Adding AmiiboLink Hardware

AndyMatt, 2023-07-03:
### Feature Explanation

I am surprised this hasn't popped up yet, I think adding the AmiiboLink that are flooding the market will be helpful for a lot of users. I am able to work around alot of the issues, but I cant imagine the average end user having a better time navigating the disjointed documentation and questionable applications. 

### Usage Confirmation

- [X] I confirm this is a request for a new feature, not an issue

AbandonedCart, 2023-07-03:
It's been requested, but there isn't an official API or even an official app for it. This will require a little more time and experimentation than is possible at the moment.

Brickleberry19, 2023-07-03:
The official offline and online app , Password: amiibo
https://amiibo.lanzoum.com/b04jvnykh

AbandonedCart, 2023-07-03:
That only seems marginally questionable. I guess we can crack that open and see how it ticks.

ottoxgam, 2023-07-04:
+1 for me! The included app is garbage and half-translated. I would much rather use TagMo if possible!

AbandonedCart, 2023-07-04:
https://github.com/boskokg/flutter_blue_plus
https://pub.dev/packages/receive_intent
https://github.com/Baseflow/flutter-permission-handler
https://github.com/florent37/Flutter-AssetsAudioPlayer
https://github.com/miguelpruivo/flutter_file_picker
https://github.com/tekartik/sqflite
https://pub.dev/packages/package_info_plus
https://pub.dev/packages/device_info_plus
https://github.com/espresso3389/flutter_content_resolver
https://github.com/perron2/async-zip

https://docs.sentry.io/platforms/android/

ottoxgam, 2023-07-04:
The app found here https://download.amiloop.app/android/ doesn't use flutter and is mostly translated.

Brickleberry19, 2023-07-04:
Amiiloop and amiiboLink are not really the same.
Amiiloop use only NFC to add an amiibo 
amiiboLink use only Bluetooth

AbandonedCart, 2023-07-05:
> 

Flutter doesn't matter. What matters is determining what is generic and what isn't. All of that is generic.

ottoxgam, 2023-07-05:
> Amiiloop and amiiboLink are not really the same. Amiiloop use only NFC to add an amiibo, thats why you can use TagMo ally etc for it. amiiboLink use only Bluetooth and works only with the omllbo/NFC tag pro app

I've got an amiloop, and it uses BLE. Both the Amiloop app I linked as well as omllbo app send the binaries to it. I can link the item if you'd like.

Brickleberry19, 2023-07-05:
Oh okay
I have the amiiboLink and Its only usable With the omllbo/NFC pro . I tried also the amiiloop App but this doesnt work for me

DanTheMan827, 2023-07-05:
It’s not complete, but there’s a basic implementation of writing from the amiloop app.

https://dantheman827.github.io/amiloop-webbluetooth/

AbandonedCart, 2023-07-05:
@AndyMatt How are you currently using the app? If it's mostly working for you, then perhaps automating some of your process would get things moving a little faster until a whole process exists.

DanTheMan827, 2023-07-08:
@AbandonedCart Here's a bluetooth capture of writing to an AmiiboLink.

Device identifies as `amiibolink`, has nordic uart services.

Appears to accept 540 byte amiibo dumps, 0xFF bytes in the sample below should be the bytes of the dump chunked at 20 bytes for each BLE packet.

```
> A0 B0
< B0 A0

> AC AC 00 04 00 00 02 1C
< CA CA

> AB AB 02 1C
< BA BA

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 01
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 02
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 03
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 04
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 05
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 06
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 07
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 08
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 09
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 0A
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 0B
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 0C
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 0D
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 0E
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 0F
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 10
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 11
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 12
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 13
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 14
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 15
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 16
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 17
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 18
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 19
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 1A
< AA DD

> DD AA 00 14 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 1B
< AA DD

> BC BC
< CB CB

> CC DD
< DD CC
```

AbandonedCart, 2023-07-08:
We've got it under control. Thanks.

AbandonedCart, 2023-08-15:
Due to concerns with branding and copyright, support will be added for the Pixl.js platform that will not endorse or list any specific device names. That being said, this issue will be closed in favor of a more generic discussion.

closed 2023-08-15 by AbandonedCart