HiddenRamblings/TagMo #716: [Issue]: Unable to connect to BluupLabs Flask

generally-disinterested, 2023-09-20:
### Issue Description

Since a recent update (unsure of which one, I didn't need to use the app for a while) I can no longer connect to my Flask. It shows up in the Bluetooth menu, but when I try to connect to it it gets stuck on "Connecting to Bluetooth GATT" and never actually connects. I have been able to connect to it via the official desktop, web, and mobile apps so I know the flask itself works, so I'm unsure why only TagMo is having issues with it.

### Logcat Output

```shell
09-20 12:31:02.634 16286 16286 E .tagmo.eightbi: Not starting debugger since process cannot load the jdwp agent.
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: Couldn't encode request, returning with 400
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: javax.net.ssl.SSLHandshakeException: java.security.cert.CertPathValidatorException: Trust anchor for certification path not found.
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.SSLUtils.toSSLHandshakeException(SSLUtils.java:362)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.ConscryptEngine.convertException(ConscryptEngine.java:1134)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.ConscryptEngine.readPlaintextData(ConscryptEngine.java:1089)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.ConscryptEngine.unwrap(ConscryptEngine.java:876)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.ConscryptEngine.unwrap(ConscryptEngine.java:747)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.ConscryptEngine.unwrap(ConscryptEngine.java:712)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.ConscryptEngineSocket$SSLInputStream.processDataFromSocket(ConscryptEngineSocket.java:849)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.ConscryptEngineSocket$SSLInputStream.access$100(ConscryptEngineSocket.java:722)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.ConscryptEngineSocket.doHandshake(ConscryptEngineSocket.java:238)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.ConscryptEngineSocket.startHandshake(ConscryptEngineSocket.java:217)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.okhttp.internal.io.RealConnection.connectTls(RealConnection.java:196)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.okhttp.internal.io.RealConnection.connectSocket(RealConnection.java:153)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.okhttp.internal.io.RealConnection.connect(RealConnection.java:116)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.okhttp.internal.http.StreamAllocation.findConnection(StreamAllocation.java:186)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.okhttp.internal.http.StreamAllocation.findHealthyConnection(StreamAllocation.java:128)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.okhttp.internal.http.StreamAllocation.newStream(StreamAllocation.java:97)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.okhttp.internal.http.HttpEngine.connect(HttpEngine.java:289)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.okhttp.internal.http.HttpEngine.sendRequest(HttpEngine.java:232)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.okhttp.internal.huc.HttpURLConnectionImpl.execute(HttpURLConnectionImpl.java:465)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.okhttp.internal.huc.HttpURLConnectionImpl.connect(HttpURLConnectionImpl.java:131)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.okhttp.internal.huc.HttpURLConnectionImpl.getOutputStream(HttpURLConnectionImpl.java:262)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.okhttp.internal.huc.DelegatingHttpsURLConnection.getOutputStream(DelegatingHttpsURLConnection.java:219)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.okhttp.internal.huc.HttpsURLConnectionImpl.getOutputStream(HttpsURLConnectionImpl.java:30)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.google.android.datatransport.cct.CctTransportBackend.doSend(CctTransportBackend.java:303)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.google.android.datatransport.cct.CctTransportBackend.$r8$lambda$bLAzIpNF4NtapXlUpPVGhzxyNT8(CctTransportBackend.java:0)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.google.android.datatransport.cct.CctTransportBackend$$ExternalSyntheticLambda0.apply(R8$$SyntheticClass:0)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.google.android.datatransport.runtime.retries.Retries.retry(Retries.java:54)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.google.android.datatransport.cct.CctTransportBackend.send(CctTransportBackend.java:372)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.google.android.datatransport.runtime.scheduling.jobscheduling.Uploader.logAndUpdateState(Uploader.java:146)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.google.android.datatransport.runtime.scheduling.jobscheduling.Uploader.lambda$upload$1(Uploader.java:105)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.google.android.datatransport.runtime.scheduling.jobscheduling.Uploader.$r8$lambda$DXUaNZ7S78mHsDrcqc_9ECz1Ymg(Uploader.java:0)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.google.android.datatransport.runtime.scheduling.jobscheduling.Uploader$$ExternalSyntheticLambda0.run(R8$$SyntheticClass:0)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.google.android.datatransport.runtime.SafeLoggingExecutor$SafeLoggingRunnable.run(SafeLoggingExecutor.java:47)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at java.lang.Thread.run(Thread.java:923)
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: Caused by: java.security.cert.CertificateException: java.security.cert.CertPathValidatorException: Trust anchor for certification path not found.
09-20 12:31:02.798 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.TrustManagerImpl.verifyChain(TrustManagerImpl.java:677)
09-20 12:31:02.799 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.TrustManagerImpl.checkTrustedRecursive(TrustManagerImpl.java:554)
09-20 12:31:02.799 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.TrustManagerImpl.checkTrusted(TrustManagerImpl.java:510)
09-20 12:31:02.799 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.TrustManagerImpl.checkTrusted(TrustManagerImpl.java:428)
09-20 12:31:02.799 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.TrustManagerImpl.getTrustedChainForServer(TrustManagerImpl.java:356)
09-20 12:31:02.799 16286 16309 E TRuntime.CctTransportBackend: at android.security.net.config.NetworkSecurityTrustManager.checkServerTrusted(NetworkSecurityTrustManager.java:94)
09-20 12:31:02.799 16286 16309 E TRuntime.CctTransportBackend: at android.security.net.config.RootTrustManager.checkServerTrusted(RootTrustManager.java:90)
09-20 12:31:02.799 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.ConscryptEngineSocket$2.checkServerTrusted(ConscryptEngineSocket.java:161)
09-20 12:31:02.799 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.Platform.checkServerTrusted(Platform.java:250)
09-20 12:31:02.799 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.ConscryptEngine.verifyCertificateChain(ConscryptEngine.java:1644)
[... cut ...]
09-20 12:32:02.590 16286 16309 E TRuntime.CctTransportBackend: at com.android.org.conscrypt.ConscryptEngine.readPlaintextData(ConscryptEngine.java:1079)
09-20 12:32:02.590 16286 16309 E TRuntime.CctTransportBackend: ... 33 more
09-20 12:32:02.590 16286 16309 E TRuntime.CctTransportBackend: Caused by: java.security.cert.CertPathValidatorException: Trust anchor for certification path not found.
09-20 12:32:02.590 16286 16309 E TRuntime.CctTransportBackend: ... 47 more
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2023-09-20:
The part of the log that was posted shows an SSL issue. Without the rest of the log, it looks like an issue with your phone not updating.

generally-disinterested, 2023-09-20:
As in the app or Android itself? If it's the former the app has been able to update fine (currently on commit 2608914, which is the most recent) but if the issue is caused by my OS not updating then I guess I'm out of luck, but I'm not sure how that would stop it from connecting to a bluetooth device when it was able to just fine until recently. As for the logs

generally-disinterested, 2023-09-20:
As in the app or Android itself? If it's the former the app has been able to update fine (currently on commit https://github.com/HiddenRamblings/TagMo/commit/26089140fb58086d6325f6af4b25483560f6dff6, which is the most recent) but if the issue is caused by my OS not updating then I guess I'm out of luck, but I'm not sure how that would stop it from connecting to a bluetooth device when it was able to just fine until recently. As for the logs, that's what was included when I clicked the "Report an Issue" button in the app, if there is any way to get more detailed logs then I can. (Apologies for closing and re-opening, accidentally clicked the wrong button when trying to submit this comment)

closed 2023-09-21 by AbandonedCart