HiddenRamblings/TagMo #717: [Issue]: Failed to decrypt tag data

122leeje, 2023-09-21:
### Issue Description

tagging any tag to write leads to the message "Failed to decrypt tag data"
it worked previously

### Logcat Output

```shell
TagMo 4.0.9 (GitHub Release) #e7fcfc1
samsung TIRAMISU (13) - 5.00 GB RAM

java.lang.Exception: Failed to decrypt tag data
	at com.hiddenramblings.tagmo.amiibo.KeyManager.decrypt(KeyManager.kt:107)
	at com.hiddenramblings.tagmo.nfctech.TagWriter.writeToTagAuto(TagWriter.kt:107)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onMifareUltralight$2.invokeSuspend(NfcActivity.kt:344)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onMifareUltralight$2.invoke(NfcActivity.kt:0)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onMifareUltralight$2.invoke(NfcActivity.kt:0)
	at kotlinx.coroutines.intrinsics.UndispatchedKt.startUndispatchedOrReturn(Undispatched.kt:78)
	at kotlinx.coroutines.BuildersKt__Builders_commonKt.withContext(Builders.common.kt:167)
	at kotlinx.coroutines.BuildersKt.withContext(Unknown Source:1)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity.onMifareUltralight(NfcActivity.kt:259)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity.access$onMifareUltralight(NfcActivity.kt:52)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onTagDiscovered$2.invokeSuspend(NfcActivity.kt:671)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onTagDiscovered$2.invoke(NfcActivity.kt:0)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onTagDiscovered$2.invoke(NfcActivity.kt:0)
	at kotlinx.coroutines.intrinsics.UndispatchedKt.startUndispatchedOrReturn(Undispatched.kt:78)
	at kotlinx.coroutines.BuildersKt__Builders_commonKt.withContext(Builders.common.kt:167)
	at kotlinx.coroutines.BuildersKt.withContext(Unknown Source:1)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity.onTagDiscovered(NfcActivity.kt:663)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity.access$onTagDiscovered(NfcActivity.kt:52)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onNewIntent$1.invokeSuspend(NfcActivity.kt:818)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.internal.LimitedDispatcher$Worker.run(LimitedDispatcher.kt:115)
	at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:100)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:584)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:793)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:697)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:684)
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2023-09-21:
Are you trying to write your own bin files?
Since you mentioned it previously working, when did it stop?

AbandonedCart, 2023-09-28:
Closing this due to lack of response.

closed 2023-09-28 by AbandonedCart