HiddenRamblings/TagMo #718: [Feature]: Use Bluetooth Connection as main Write action.

ishimarumakoto, 2023-09-22:
### Feature Explanation

As A user I would like to use my BT device as the main device instead of creating a NFC tag.

I would like to click on write in the main page and if connected to BT device, send the bin seemlessly to device.

### Usage Confirmation

- [X] I confirm this is a request for a new feature, not an issue

AbandonedCart, 2023-09-22:
Given the process to set up a Bluetooth device, this would cause a lot of confusion and trouble for most. That aside, amiibo are still NFC tags. Bluetooth support is a secondary feature.

closed 2023-09-22 by AbandonedCart

AbandonedCart, 2023-09-22:
I realized this may have come off a bit dismissive because I was half awake writing it. The process to connect a device is multiple steps, many of which are repeated throughout the connection. To actually do something like this, it would end up being essentially the same as connecting to the device and selecting write. The only difference would be that you chose the amiibo and then write over write and then the amiibo. The entire connection process would still be required the first time to ensure a write is possible. 
The problem with setting to write to optionally use Bluetooth is that you could easily end up overwriting a Bluetooth device intending to write to NFC. Adding a confirmation will only lengthen the process across the board and I assume the point is to simplify it.
NFC is still the primary function, so it has to be assumed that most people wanted to write NFC. They shouldn't have to confirm they aren't writing to Bluetooth for that.

ishimarumakoto, 2023-09-22:
> I realized this may have come off a bit dismissive because I was half awake writing it. The process to connect a device is multiple steps, many of which are repeated throughout the connection. To actually do something like this, it would end up being essentially the same as connecting to the device and selecting write. The only difference would be that you chose the amiibo and then write over write and then the amiibo. The entire connection process would still be required the first time to ensure a write is possible. The problem with setting write to optionally use Bluetooth is that you could easily end up overwriting a Bluetooth device intending to write to NFC. Adding a confirmation will only lengthen the process across the board and I assume the point is to simplify it. NFC is still the primary function, so it has to be assumed that most people wanted to write NFC. They shouldn't have to confirm they aren't writing to Bluetooth for that.

Thank you for taking your time to explain :)
I understand your point, I think then a select option of your primary way of writing would help, but I agree that is simple enough the way it is now, it was just a "shortcut" for us with bt devices.
Cheers and congratulations on the project

AbandonedCart, 2023-09-22:
I am looking into some compromises, such as a separate upload button. First, though, the menus really need some cleanup. The number of options is getting a bit unmanageable.