HiddenRamblings/TagMo #720: [Issue]: Cant Back Up Amiibolink

githubUser01946, 2023-10-04:
### Issue Description

I have scanned my amiibolink and it chosen backup and it shows there are:

- 638 characters
- 4 amiibo types
- 102 enabled games

But I don't know how to view them. Update document storage shows an empty folder, extract bins from zip just opens the folder and does nothing.

How do I see what is on my Amiibolink in order to back it up?

### Logcat Output

```shell
TagMo 4.1.1 (Play Release) #7441e031
samsung TIRAMISU (13) - 7.00 GB RAM
KeyManager Status: Key(s) Missing

--------- beginning of main
10-04 17:58:35.180 12503 12550 E libEGL  : eglSetBlobCachePathSEC  resulted in an error: 0x300c
10-04 17:58:35.191 12503 12550 W vulkan  : CreateSwapchainKHR(VkSwapchainCreateInfoKHR.imageColorSpace = 1000104013) failed: Unsupported color space, but assume as Success for ANGLE.
10-04 17:58:50.649 12503 12550 E libEGL  : eglSetBlobCachePathSEC  resulted in an error: 0x300c
10-04 17:58:50.653 12503 12550 W vulkan  : CreateSwapchainKHR(VkSwapchainCreateInfoKHR.imageColorSpace = 1000104013) failed: Unsupported color space, but assume as Success for ANGLE.
10-04 18:00:35.508 12503 12550 W vulkan  : CreateSwapchainKHR(VkSwapchainCreateInfoKHR.imageColorSpace = 1000104013) failed: Unsupported color space, but assume as Success for ANGLE.
10-04 18:01:14.819 12503 12550 W vulkan  : CreateSwapchainKHR(VkSwapchainCreateInfoKHR.imageColorSpace = 1000104013) failed: Unsupported color space, but assume as Success for ANGLE.
10-04 18:01:14.841 12503 12550 W Parcel  : Expecting binder but got null!
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2023-10-04:
Sounds like you're combining a couple different sections together. Looking over the instructions may help to sort things out. https://tagmo.gitlab.io/

Special instructions for the device are in the key fob section. Backup works like any other tag or amiibo, so that's on the main page.

AbandonedCart, 2023-10-05:
No screenshot is included and the backup instructions are on the main page, as listed in the quoted text.

githubUser01946, 2023-10-05:
Apologies, the screenshot must've been lost in the ether.
I don't see a backup option at all on the main page unless you are referring to "Cloning" or will it be the one before, "saving". 

As you can see in the now uploaded screenshot it has more than one amiibo on it and I want to see what they are and keep them stored in the folder I have named as Amiibo.

AbandonedCart, 2023-10-05:
You're sliding up the the bottom bar. Scan the tag first.

AbandonedCart, 2023-10-05:
Updated the guide to specifically use the word "backup" alongside saving. Cloning is also a form of backup, but you write to a tag in the final step, instead of saving a file.

closed 2023-10-05 by AbandonedCart