HiddenRamblings/TagMo #721: Unable to load N2 Elite list

Sora1995, 2023-11-05:
I managed to get the current amiibo to display when scanning the N2, but the list will not load when under the N2 menu. Logcat below, if it helps. May not though. 

TagMo 4.1.1 (GitHub Release) #96f1636
Google UPSIDE_DOWN_CAKE (14) - 12.00 GB RAM

--------- beginning of main
11-05 13:45:22.919 29017 29017 W View    : requestLayout() improperly called by android.widget.FrameLayout{b61ac03 V.E..V... ......ID 0,0-1440,2612 #2} during layout: running second layout pass
11-05 13:45:22.920 29017 29017 W View    : requestLayout() improperly called by androidx.recyclerview.widget.RecyclerView{7749f3f IFED.V... ......I. 0,2626-1440,2626 #7f090108 app:id/foomiibo_list} during layout: running second layout pass
11-05 13:45:22.920 29017 29017 W View    : requestLayout() improperly called by androidx.recyclerview.widget.RecyclerView{3ba0412 VFED.V... ......ID 0,0-1440,2608 #7f090081 app:id/browser_content} during layout: running second layout pass
11-05 13:45:39.021 29017 29017 W WindowOnBackDispatcher: sendCancelIfRunning: isInProgress=falsecallback=androidx.activity.OnBackPressedDispatcher$Api33Impl$$ExternalSyntheticLambda0@7fa0c6b
11-05 13:45:55.177 29017 29017 W Choreographer: Frame time is 0.110346 ms in the future!  Check that graphics HAL is generating vsync timestamps using the correct timebase.
11-05 13:46:23.795 29017 29017 W WindowOnBackDispatcher: sendCancelIfRunning: isInProgress=falsecallback=android.widget.PopupWindow$PopupDecorView$$ExternalSyntheticLambda0@e36fb4e

Sora1995, 2023-11-05:
Oddly, clicking "Scan from Tag" managed to load the entire list. Not sure if this is intentional. Disregard report if it is. Thanks!

AbandonedCart, 2024-05-08:
Appears I never responded, but work was done to resolve this issue when it was originally posted.

closed 2024-05-08 by AbandonedCart