HiddenRamblings/TagMo #722: [Issue]: bluetooth connection problem with Amiloop

Vorffeed, 2023-11-17:
### Issue Description

As much as I try to pair the amiilop, all the time it refuses the connection is there any way to connect it to be able to write on the amiibos on the device?

### Logcat Output

```shell
As much as I try to pair the amiilop, all the time it refuses the connection is there any way to connect it to be able to write on the amiibos on the device?
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2023-11-18:
The instructions for connecting are posted at https://tagmo.gitlab.io/keyfobs.html

closed 2023-11-18 by AbandonedCart

Vorffeed, 2023-11-18:
I have looked in the instructions it says that you can connect, but I get an error I have a Redmi note 9 pro, with other app I have no problems to connect the AmiLoop and when I try to connect I get this 
[image]
[image]

and it does not connect, I have tried to install it on my mother's phone which is a redmi note 9 and she connects without problems ><

I have used lodgcat reader to see if I could find the bug, I don't know if it will be in this piece because it shows me a lot of information and I don't understand how to make it monitor only the tagmo ><

`:31:33.090  3740  3740 I KeyguardIndicationInjector: handleChargeTextAnimation: false;powerPluggedIn=true
11-18 12:31:33.097  1676  5186 I WindowManagerService: Physical orientation not applied, rotation=0
11-18 12:31:33.248  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:33.448  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:33.650  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:33.851  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:33.876  1676  2007 I BatteryService: current mPlugType1current mHealthInfo.batteryLevel40current mHealthInfo.batteryTemperature372
11-18 12:31:33.880  5695  5695 I BatteryInfoReceiver: ACTION_BATTERY_CHANGED
11-18 12:31:33.886  5695  5695 I PowerNoticeUI: handleBatteryChanged oldPlugType 1 mPlugType 1 oldBatteryLevel 40 mBatteryLevel 40 temperature 372
11-18 12:31:34.051  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:34.130  1676  7576 I WindowManagerService: Physical orientation not applied, rotation=0
11-18 12:31:34.252  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:34.453  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:34.456  1676  7576 W InputManager-JNI: Input channel object '3fd1ede PopupWindow:a8515f1 (client)' was disposed without first being removed with the input manager!
11-18 12:31:34.653  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:34.670  3384  3826 I BtGatt.ScanManager: msg.what = 6
11-18 12:31:34.670  1676  3270 I MiuiNetworkPolicy: updateUidState uid = 10196, uidState = 5
11-18 12:31:34.854  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:35.055  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:35.101  1676  7576 W InputManager-JNI: Input channel object '878ec5 com.hiddenramblings.tagmo.eightbit/com.hiddenramblings.tagmo.BrowserActivity (client)' was disposed without first being removed with the input manager!
11-18 12:31:35.178  3384  4066 I bt_stack: [INFO:gatt_api.cc(1239)] GATT_Register 7cba8101-ce13-4f4b-6e6b-8f0a38a7cf80
11-18 12:31:35.178  3384  4066 I bt_stack: [INFO:gatt_api.cc(1260)] allocated gatt_if=10
11-18 12:31:35.185  3384  4066 I bt_stack: [INFO:gatt_api.cc(1429)] GATT_Connect gatt_if=10, address=e4:a2:5d:9f:01:d9 is_direct 1
11-18 12:31:35.187  1043  1043 I vendor.qti.bluetooth@1.0-uart_transport: ## userial_vendor_ioctl: UART Flow Off
11-18 12:31:35.188  1043  1043 I vendor.qti.bluetooth@1.0-uart_transport: ## userial_vendor_set_baud: 17
11-18 12:31:35.189  1043  1043 I vendor.qti.bluetooth@1.0-uart_transport: SetBaudRate: in BOTHER
11-18 12:31:35.189  1043  1043 I vendor.qti.bluetooth@1.0-uart_transport: ## userial_vendor_ioctl: UART Flow On
11-18 12:31:35.189  1043  1043 I vendor.qti.bluetooth@1.0-ibs_handler: DeviceWakeUp: Writing IBS_WAKE_IND
11-18 12:31:35.191  1043  4063 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_ACK: 0xFC
11-18 12:31:35.191  1043  4063 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Signal wack_cond_
11-18 12:31:35.193  1043  4063 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
11-18 12:31:35.193  1043  4063 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
11-18 12:31:35.210 16324 16324 E FullInputEventModel: onStartInput event aborted: ky.y: could not obtain extracted text (class ky.y)
11-18 12:31:35.256  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:35.274  1043  4063 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
11-18 12:31:35.321  1043  4063 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
11-18 12:31:35.322  1043  4063 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
11-18 12:31:35.324  3384  4066 I bt_stack: [INFO:btm_ble_connection_establishment.cc(179)] btm_ble_conn_complete BD addr e4:a2:5d:9f:01:d9 Not resolved
11-18 12:31:35.324  3384  4066 I bt_stack: [INFO:connection_manager.cc(280)] lower the scan parameters used for connecting
11-18 12:31:35.325  3384  4066 I bt_stack: [INFO:btm_ble_bgconn.cc(429)] Already connection initiated (or) No Pending connections(or) Resources of lcb are exhausted
11-18 12:31:35.325  3384  4066 W bt_btm  : btm_acl_created hci_handle=12 link_role=0  transport=2
11-18 12:31:35.325  3384  4066 I bt_stack: [INFO:btm_ble_bgconn.cc(429)] Already connection initiated (or) No Pending connections(or) Resources of lcb are exhausted
11-18 12:31:35.333  3384  4066 I bt_stack: [INFO:gatt_main.cc(1630)] gatt_data_process op_code = 2, msg_len = 2
[... cut ...]
11-18 12:31:35.461  1676  2027 I ActivityManager: Start proc 29227:com.google.android.projection.gearhead:shared/u0a198 for broadcast {com.google.android.projection.gearhead/com.google.android.apps.auto.wireless.bluetooth.WifiBluetoothReceiver} caller=com.android.bluetooth
11-18 12:31:35.497 29227 29227 I gearhead:share: Using CollectorTypeCC GC.
11-18 12:31:35.511  3384  4066 I bt_stack: [INFO:gatt_main.cc(1630)] gatt_data_process op_code = 17, msg_len = 13
11-18 12:31:35.511  3384  4066 W bt_stack: [WARNING:gatt_cl.cc(1288)] ATT - clcb already not in use, ignoring response`