HiddenRamblings/TagMo #723: [Issue]: Amiloop rejects bluetooth connection

Vorffeed, 2023-11-18:
### Issue Description

Perdón por repetirme, pero es que el otro lo han dado como caso cerrado.
I have looked in the instructions it says that you can connect, but it gives me error I have a Redmi note 9 pro, with another app I have no problem connecting the AmiLoop and when I try to connect from Tagmo to Amiloop mes comes out this 

[image]
[image]

And it does not connect, I have tried to install it on my mother's phone which is a redmi note 9 and she connects without problems ><

I have used lodgcat reader to see if I could find the bug, I don't know if it will be in this piece because it shows me a lot of information and I don't understand how to make it monitor only the tagmo ><

### Logcat Output

```shell
11-18 12:31:33.090  3740  3740 I KeyguardIndicationInjector: handleChargeTextAnimation: false;powerPluggedIn=true
11-18 12:31:33.097  1676  5186 I WindowManagerService: Physical orientation not applied, rotation=0
11-18 12:31:33.248  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:33.448  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:33.650  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:33.851  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:33.876  1676  2007 I BatteryService: current mPlugType1current mHealthInfo.batteryLevel40current mHealthInfo.batteryTemperature372
11-18 12:31:33.880  5695  5695 I BatteryInfoReceiver: ACTION_BATTERY_CHANGED
11-18 12:31:33.886  5695  5695 I PowerNoticeUI: handleBatteryChanged oldPlugType 1 mPlugType 1 oldBatteryLevel 40 mBatteryLevel 40 temperature 372
11-18 12:31:34.051  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:34.130  1676  7576 I WindowManagerService: Physical orientation not applied, rotation=0
11-18 12:31:34.252  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:34.453  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:34.456  1676  7576 W InputManager-JNI: Input channel object '3fd1ede PopupWindow:a8515f1 (client)' was disposed without first being removed with the input manager!
11-18 12:31:34.653  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:34.670  3384  3826 I BtGatt.ScanManager: msg.what = 6
11-18 12:31:34.670  1676  3270 I MiuiNetworkPolicy: updateUidState uid = 10196, uidState = 5
11-18 12:31:34.854  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:35.055  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:35.101  1676  7576 W InputManager-JNI: Input channel object '878ec5 com.hiddenramblings.tagmo.eightbit/com.hiddenramblings.tagmo.BrowserActivity (client)' was disposed without first being removed with the input manager!
11-18 12:31:35.178  3384  4066 I bt_stack: [INFO:gatt_api.cc(1239)] GATT_Register 7cba8101-ce13-4f4b-6e6b-8f0a38a7cf80
11-18 12:31:35.178  3384  4066 I bt_stack: [INFO:gatt_api.cc(1260)] allocated gatt_if=10
11-18 12:31:35.185  3384  4066 I bt_stack: [INFO:gatt_api.cc(1429)] GATT_Connect gatt_if=10, address=e4:a2:5d:9f:01:d9 is_direct 1
11-18 12:31:35.187  1043  1043 I vendor.qti.bluetooth@1.0-uart_transport: ## userial_vendor_ioctl: UART Flow Off
11-18 12:31:35.188  1043  1043 I vendor.qti.bluetooth@1.0-uart_transport: ## userial_vendor_set_baud: 17
11-18 12:31:35.189  1043  1043 I vendor.qti.bluetooth@1.0-uart_transport: SetBaudRate: in BOTHER
11-18 12:31:35.189  1043  1043 I vendor.qti.bluetooth@1.0-uart_transport: ## userial_vendor_ioctl: UART Flow On
11-18 12:31:35.189  1043  1043 I vendor.qti.bluetooth@1.0-ibs_handler: DeviceWakeUp: Writing IBS_WAKE_IND
11-18 12:31:35.191  1043  4063 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_ACK: 0xFC
11-18 12:31:35.191  1043  4063 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Signal wack_cond_
11-18 12:31:35.193  1043  4063 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
11-18 12:31:35.193  1043  4063 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
11-18 12:31:35.210 16324 16324 E FullInputEventModel: onStartInput event aborted: ky.y: could not obtain extracted text (class ky.y)
11-18 12:31:35.256  3694  8700 I SarService: Receiver polling >>>> Change receiver mode: off
11-18 12:31:35.274  1043  4063 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
11-18 12:31:35.321  1043  4063 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
11-18 12:31:35.322  1043  4063 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
11-18 12:31:35.324  3384  4066 I bt_stack: [INFO:btm_ble_connection_establishment.cc(179)] btm_ble_conn_complete BD addr e4:a2:5d:9f:01:d9 Not resolved
11-18 12:31:35.324  3384  4066 I bt_stack: [INFO:connection_manager.cc(280)] lower the scan parameters used for connecting
11-18 12:31:35.325  3384  4066 I bt_stack: [INFO:btm_ble_bgconn.cc(429)] Already connection initiated (or) No Pending connections(or) Resources of lcb are exhausted
11-18 12:31:35.325  3384  4066 W bt_btm  : btm_acl_created hci_handle=12 link_role=0  transport=2
11-18 12:31:35.325  3384  4066 I bt_stack: [INFO:btm_ble_bgconn.cc(429)] Already connection initiated (or) No Pending connections(or) Resources of lcb are exhausted
11-18 12:31:35.333  3384  4066 I bt_stack: [INFO:gatt_main.cc(1630)] gatt_data_process op_code = 2, msg_len = 2
11-18 12:31:35.333  3384  4066 I bt_stack: [INFO:gatt_sr.cc(915)] MTU request PDU with MTU size 247
11-18 12:31:35.333  3384  4066 E bt_btm  : BTM_SetBleDataLength failed, peer does not support request
11-18 12:31:35.334  3384  4066 W bt_stack: [WARNING:gatt_utils.cc(916)] Call back not found for application conn_id=3
11-18 12:31:35.334  3384  4066 W bt_stack: [WARNING:gatt_utils.cc(916)] Call back not found for application conn_id=4
11-18 12:31:35.334  3384  4066 W bt_stack: [WARNING:gatt_utils.cc(916)] Call back not found for application conn_id=5
11-18 12:31:35.334  3384  4066 W bt_stack: [WARNING:gatt_utils.cc(916)] Call back not found for application conn_id=7
11-18 12:31:35.334  3384  4066 W bt_stack: [WARNING:gatt_utils.cc(916)] Call back not found for application conn_id=8
11-18 12:31:35.335  3384  4066 W bt_stack: [WARNING:gatt_utils.cc(916)] Call back not found for application conn_id=9
11-18 12:31:35.335  3384  4066 W bt_stack: [WARNING:gatt_utils.cc(916)] Call back not found for application conn_id=10
11-18 12:31:35.335  3384  4066 E bt_btif : bta_dm_acl_change new acl connetion:count = 1
11-18 12:31:35.335  3384  4066 W bt_btif : bta_dm_acl_change info: 0x0
11-18 12:31:35.335  3384  4066 W bt_l2cap: L2CA_SetDesireRole() new:x1, disallow_switch:0
[... 35 lines cut ...]
11-18 12:31:35.455 28898 28898 W IllegalAccessException: at com.hiddenramblings.tagmo.fragment.GattSlotFragment$onViewCreated$2$2$1.onItemSelected(GattSlotFragment.kt:183)
11-18 12:31:35.455 28898 28898 W IllegalAccessException: at android.widget.AdapterView.fireOnSelected(AdapterView.java:957)
11-18 12:31:35.455 28898 28898 W IllegalAccessException: at android.widget.AdapterView.dispatchOnItemSelected(AdapterView.java:946)
11-18 12:31:35.455 28898 28898 W IllegalAccessException: at android.widget.AdapterView.access$300(AdapterView.java:55)
11-18 12:31:35.455 28898 28898 W IllegalAccessException: at android.widget.AdapterView$SelectionNotifier.run(AdapterView.java:910)
11-18 12:31:35.455 28898 28898 W IllegalAccessException: at android.os.Handler.handleCallback(Handler.java:938)
11-18 12:31:35.455 28898 28898 W IllegalAccessException: at android.os.Handler.dispatchMessage(Handler.java:99)
11-18 12:31:35.455 28898 28898 W IllegalAccessException: at android.os.Looper.loopOnce(Looper.java:210)
11-18 12:31:35.455 28898 28898 W IllegalAccessException: at android.os.Looper.loop(Looper.java:299)
11-18 12:31:35.455 28898 28898 W IllegalAccessException: at android.app.ActivityThread.main(ActivityThread.java:8319)
11-18 12:31:35.455 28898 28898 W IllegalAccessException: at java.lang.reflect.Method.invoke(Native Method)
11-18 12:31:35.455 28898 28898 W IllegalAccessException: at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:556)
11-18 12:31:35.455 28898 28898 W IllegalAccessException: at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1038)
[... cut ...]
11-18 12:31:35.461  1676  2027 I ActivityManager: Start proc 29227:com.google.android.projection.gearhead:shared/u0a198 for broadcast {com.google.android.projection.gearhead/com.google.android.apps.auto.wireless.bluetooth.WifiBluetoothReceiver} caller=com.android.bluetooth
11-18 12:31:35.497 29227 29227 I gearhead:share: Using CollectorTypeCC GC.
11-18 12:31:35.511  3384  4066 I bt_stack: [INFO:gatt_main.cc(1630)] gatt_data_process op_code = 17, msg_len = 13
11-18 12:31:35.511  3384  4066 W bt_stack: [WARNING:gatt_cl.cc(1288)] ATT - clcb already not in use, ignoring response
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2023-11-18:
There is a known issue with the Bluetooth for some devices manufactured in China (Xiaomi, OPPO, OnePlus). The failure happens in the drivers, so we are still investigating a workaround.

Vorffeed, 2023-11-18:
You mean the Tagmo controller?
Because with others app connects perfectly to Amiloop on my mobile only happens to me with Tagmo that gives that problem, but this is the only app that has everything I'm looking for to transfer and dump the content back from Amiloop to mobile, the others only send and I can not recover the Amiibo when the data of this is modified as happens in Super smash Bros that saves the progress of the Amiibo to put back the Amiloop to transfer the data to the Amiibo, and if I inject another Amiibo these data disappear to not being able to make the dump and save them.😭

AbandonedCart, 2023-11-18:
The issue is with the Bluetooth software on the phone. TagMo needs a workaround for an issue with the Bluetooth software built into the phone..

Vorffeed, 2023-11-19:
Sorry, my intention is not to offend or anything like that, I speak from the lack of knowledge of how all this works and as I use translator, I don't know if it translates well what you say or I say ^^U.
I don't understand very well, what you mean is that it's a problem of the phone software?
Then because I can without problems with other apps and I can send the amiibos to Amiloop from the mobile is what has surprised me the most.

I do not know if this can help you, so that it can connect I have taken it out of the app Nrf Connect to pair it with the default AmiLoop application of the keychain, I only use it with this app, the other as Umiibo connects itself directly without using anything additional.  
[image]
[image]
[image]
[image]

```nRF Connect, 2023-11-19
AmiLoop (E4:A2:5D:9F:01:D9)
V	12:54:58.957	Connecting to E4:A2:5D:9F:01:D9...
D	12:54:58.957	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	12:54:59.291	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	12:54:59.296	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	12:54:59.296	Connected to E4:A2:5D:9F:01:D9
I	12:54:59.312	MTU changed to: 247
V	12:54:59.329	Discovering services...
D	12:54:59.329	gatt.discoverServices()
V	12:54:59.336	Discovering services...
D	12:54:59.336	gatt.discoverServices()
I	12:54:59.955	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	12:55:00.097	[Callback] Services discovered with status: 0
I	12:55:00.097	Services discovered
V	12:55:00.115	Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
- Central Address Resolution [R] (0x2AA6)
Generic Attribute (0x1801)
Nordic UART Service (6e400001-b5a3-f393-e0a9-e50e24dcca9e)
- RX Characteristic [W WNR] (6e400002-b5a3-f393-e0a9-e50e24dcca9e)
- TX Characteristic [N] (6e400003-b5a3-f393-e0a9-e50e24dcca9e)
   Client Characteristic Configuration (0x2902)
Secure DFU Service (0xFE59)
- Buttonless DFU [I W] (8ec90003-f315-4f60-9fb8-838830daea50)
   Client Characteristic Configuration (0x2902)
D	12:55:00.116	gatt.setCharacteristicNotification(6e400003-b5a3-f393-e0a9-e50e24dcca9e, true)
I	12:55:00.175	Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)
I	12:55:05.313	PHY updated (TX: LE 2M, RX: LE 2M)
I	12:55:05.772	Connection parameters updated (interval: 75.0ms, latency: 0, timeout: 4000ms)
```

AbandonedCart, 2023-11-19:
If another app works, your best bet would be to use that app. It is still unknown why TagMo won't connect on some phones and the apps that connect do not share their source.

Vorffeed, 2023-11-19:
Thank you very much with this new update has been connected.😍

closed 2023-11-20 by AbandonedCart