HiddenRamblings/TagMo #724: [Feature]: Amiibo .bin update button

Vorffeed, 2023-11-20:
### Feature Explanation

Could you implement a button to be able to update the keychain .bin file in the .bin of the amiibo of the mobile?

I explain to have if I understand what I mean, I know there is the function to create Backup, but it is not what I mean, because in this way you find yourself with 400 .bin files on the mobile of the same amiibo.

I mean to be able to select the amiibo in the application and transfer the data from the keychain to this updating it, why I say this, I say this because there are several games like Super Smash Bros where we can save the progress of our amiibo in the keychain and others that right now I do not remember what they were that you could save the games of the game in the amiibo.

The only way I have seen for now is to create backup of the amiibo in the keychain so as not to lose the data of the progress of leveling up in Super Smash Bros or the saved games, without losing them every time we change amiibo in the keychain, but in this way we can find 400 copies of the same amiibo and have to go deleting the others each time it is created, if you don't forget to delete them ^^U, when with update the same amiibo in the mobile we can recover it later without finding duplicated amiibo and without knowing which copy is the one with the last updated progress ^^U   

### Usage Confirmation

- [X] I confirm this is a request for a new feature, not an issue

AbandonedCart, 2023-11-20:
Unfortunately, the key fob devices didn't really bother with a way to read what's already on them. TagMo doesn't overwrite existing backups because some people do want multiple copies and, in theory, every instance of the file would be slightly different. This leaves no way to confirm it's actually the same file. If you attempt to write a different amiibo back to a tag, it will break the tag.

Vorffeed, 2023-11-20:
What I was referring to is the possibility of being able to replace that .bin without generating a new copy, for those who want to update that amiibo with new data without having multiple copies of the same amiibo, I do not mean to remove the function of creating copies, but to add a new function that replaces the existing copy in the current keychain with the new data.
That is to say to make a substitution in the .bin of the amiibo of the mobile, since it does not affect the tag, since this is written the same as the copy is saved in the mobile and then rewritten in the keychain or in the tag, because this is basically to have the same id of the amiibo osea without making random id in the keychain because the same console already tells you that it does not match and does not let you update in that amiibo because it tells you that it has another id that does not match. 

I hope I explained well

That is to say to have at the same time to create backup for those who want multiple backups and the option to update / replace the bin file of the amiibo for those who do not want multiple copies.

AbandonedCart, 2023-11-20:
What I was saying is that to "update" a file, it would require scanning all of your files, finding every one that is the "same" amiibo, determining which one to replace, then replacing that file. Every step creates a little more risk for mistakes. 

When you write an NFC tag, it locks certain items. If you attempt to write one that doesn't have matching values for those locked items, it will either fail or break the tag. This is why it matters to make sure you are replacing the right file. It may not matter for you with a Bluetooth device, but TagMo is first and foremost an NFC app.

Vorffeed, 2023-11-20:
I think we are not understanding each other ^^U

One thing doesn't take away the other, I know that Tagmo was designed for tag cards, nobody argues it and I'm very grateful that they have given support for keychains.

What you say that the tag is broken I don't understand it because I think that those who use the tag cards record the data from Foomiibo Panel and they can't modify it, that is to say they record from there to the tag and then they make their backup copy that appear in Backup Browser and if they want then from there they can write to new tags.

Now what I mean by update is from Backup Browser, because there I have all my .bin files of the original Amiibos I have and these are in the memory of the mobile device, and it is from where you create your own backup of your Amiibos and these are saved, are those files that I say to be able to update with the new data that the console when writing in the keychain, those data if you have not generated a new ID by pressing the button is the same ID.

Well and although a new ID has been generated remains the same Amiibo with new data that put the console, we have not changed the essence of that Amiibo, ie, if I put a Pikachu and press 30 times will still be a Pikachu but with different serial which I can backup and put it back on the keychain or a new label without problems.

That's why I told you before that one thing did not remove the other from the functionality of creating backup and update, and I do not understand what you say about doing searches if you already have selected the .bin of the amiibo that is in your internal memory of the mobile and is your own file to modify it would only be update or overwrite it with the same name and not have in Backup Browser from 3 to 10 amiibos equal to know what modification has each one because they all have the same name on the screen ^^U

You only have one amiibo that you can modify and recover every time you need it, without having more amiibos of the account in the mobile folder where they are saved.
And if by some mistake or whatever it gets corrupted you have Foomiibo to put it back in the internal memory of the mobile or restore it from the PC copy.

I hope I explained ^_^

AbandonedCart, 2023-11-20:
I will not be adding this feature.

closed 2023-11-20 by AbandonedCart

Vorffeed, 2023-11-20:
No problem, it was just a suggestion to avoid having multiple copies of amiibos ^_^

AbandonedCart, 2023-11-20:
A simple solution would be to delete the file you are replacing when you replace it, rather than allowing them to pile up. An option to delete a file is provided.