HiddenRamblings/TagMo #729: Allow writing of non-readonly amiibo tags

Golem642, 2024-03-29:
### Feature Explanation

Hello, recently got a pack of Ntag215 and i tried writing an amiibo to one of them. Then I tried overwriting for a new Amiibo model but it didn't work, so i searched and i found out you can't rewrite amiibo serial and such because the 3DS would check for the read-only bytes as said in #18 

So i decided to make an IPS patch for it (gonna publish it right after, I'm on phone for now) so that it wouldn't check those bytes when reading/writing.
So now I'm asking : can you implement a feature so we can write amiibos without locking them to read-only ?

That would be very useful, thx

### Usage Confirmation

- [X] I confirm this is a request for a new feature, not an issue

AbandonedCart, 2024-03-29:
This would result in a mountain of issues from people not only confused how to use when they had all the right parts, but those thinking they can do it without them.

Golem642, 2024-03-29:
> This would result in a mountain of issues from people not only confused how to use when they had all the right parts, but those thinking they can do it without them.

Just put a warning or something that tells them that if they want to change the amiibo type they need a special patch for their 3DS and WiiU

Can't do the switch patch cuz I don't have one, the 3DS patch is done and OG amiibo tags can still be recognized so I'll consider this as a win (can't try with read unlocked ones since i don't know how to make one manually), WiiU patch needs to be made but i don't have access to it rn so later i guess

AbandonedCart, 2024-03-29:
> Just put a warning

The mountain of issues I was talking about was **with** a warning.

Golem642, 2024-03-29:
Like when you want to write an amiibo Tag, on the NFC scan page put a red text somewhere on the bottom that says "Amiibo tags cannot be rewritten, if you want to have write-unlocked tags you need to first install these patches for your console"

And then idk if you enable an option in the settings of the app or something it will write write-unlocked amiibo tags, and the text will be blue and say instead "Currently writing a write-unlocked amiibo tag. These will only be compatible with a modded console and the following patches installed"

Just an idea, but yk it's possible and it would help people a ton imo

AbandonedCart, 2024-03-29:
I take it you haven't actually used TagMo before. You're basically describing how loading keys works. It's the number one topic in reports.

Golem642, 2024-03-29:
> You're basically describing how loading keys works.

No ? It is the first time I'm actually using the app yeah, before that it didn't work with my other tags
But i don't get what you mean, what does it have to do with loading keys ? I'm talking about the page where you scan to write the NFC tag

AbandonedCart, 2024-03-29:
> Just put a warning or something
> put a red text somewhere on the bottom
> if you enable an option in the settings of the app or something it will write write-unlocked amiibo tags, and the text will be blue

Change the text from blue to green and you just described keys, but it's still the number one thing reported as "broken"

AbandonedCart, 2024-03-29:
I'll look into it, but I'm not making any promises. It may need to be its own separate build not available on Google Play

Golem642, 2024-03-29:
Oh this, yeah i know what you mean. I figured it out by myself but i guess not everyone does.
I have to say : the file picker opening as soon as you first start the app isn't very intuitive, like what do we have to do with it ?
That's why people don't find it, then they have to check in the settings but people just want things to work.
You could fix this by adding a dialog box **before** opening the file picker automatically, one that says "Important information: To decrypt amiibo tags, you must import unfixed-info.bin and locked-secret.bin. How you obtain the files isn't my problem but they are required for everything"

AbandonedCart, 2024-03-29:
I'm sure the app needs more dialogs. Nobody has issues figuring out how to load the keys. They don't understand why they need them. Similarly, the issue with your proposal is that most people aren't going to want to install your app and will want it on Switch.

Golem642, 2024-03-29:
I mean it could probably be implemented on switch as well, i just personnally can't do it since i don't own one. But i'll detail the patch in my repo so it's easy enough to understand

AbandonedCart, 2024-03-29:
It's nice to have a solution for everything, but I can't really implement any of it from here. so I'll get back to you.

AbandonedCart, 2024-04-01:
https://github.com/HiddenRamblings/TagMo/commit/92a0be73ec1e64321cd3c7c0fb48a49bbe3d6c9e

Is the the only part you need removed for it to work with your patch?

AbandonedCart, 2024-04-01:
Alright. I looked further into it and found a compromise for making it "safe" by using something like the current "ignore tag ID" checkbox that is opted into manually before a write. Ironically, I think that has a somewhat similar purpose. it's basically a force overwrite.

Golem642, 2024-05-05:
Good news : after a month of not having anough time to think about this and a bit more tinkering, i found a working solution !
Currently the patch can only be manually applied by editing the values in the memory using Rosalina's process manager, but i can confirm it allows reading of non write-locked nfc tags
Gonna test the writing, my excitement is kinda high rn (second time using ghidra, first time doing something useful out of it)

Golem642, 2024-05-08:
I have a question : since the point of this feature is to have rewritable tags, would it be possible to have a "Reset NTAG" kind of thing ? Like it would completely empty an NTAT215 and put it back to factory default or whatever
I know amiibos inner data are write-locked by a key generated with "unfixed infos" and "locked secret", but idk how to calculate it so i can't do it myself. Do you have a documentation on how to do that ?
It would be nice if say i want to use an old ntag for something else than being an amiibo

AbandonedCart, 2024-05-08:
https://gbatemp.net/threads/diy-amiibo-cards.406978/

That is essentially all the documentation, along with anything that can be taken from 

https://github.com/socram8888/amiitool

Golem642, 2024-06-07:
Ok so everything works for 3DS/2DS. The patch applies correctly and you can read both normal and unlocked tags
I made a release in my repo, and inside i also included a version of TagMo where i commented the lock bits parts so it can write unlocked tags.
Does this bother you or can i leave it in ?

AbandonedCart, 2024-06-07:
If you feel that's necessary and don't want it added to the main version, that's fine.

Golem642, 2024-06-07:
No I would want it to be included in the main version, it's just that this is kind of a working temporary solution I give out

AbandonedCart, 2024-06-13:
A new version was just published, so as soon as that gets approved, I can add the UI to support this one.

AbandonedCart, 2024-08-10:
It took way longer than expected, partially due to some personal issues, but I finally set everything up so all that's missing now is a button to activate it. For now, it's a per-write option, but that can easily be turned into a persistent setting.

Golem642, 2024-08-10:
No worries, take your time.
If you have a bit more time, it would be great to perform a check for unlocked tags when writing for tags **with** the lock to avoid accidental locking and losing a rewritable tag (trying to rewrite a locked tag doesn't need a check, it returns an error when attempting to write without crashing the app).
In any case, thank you for taking the time for this feature !

AbandonedCart, 2024-08-10:
That makes sense, especially if it's something that needs to be enabled. I'll have to figure out the logic to avoid confusing either side. It almost seems like the best option would be to flat out prompt "Would you like to lock the tag?" with a quick explanation on each write.

AbandonedCart, 2024-12-09:
As of the latest build, there is now a checkbox for this when writing and updating. In the three dot (context) menu, it's "Skip Lock Info" but it has to be manually checked at the moment.

Golem642, 2024-12-09:
Thank you, I will be testing it later.
I also noticed that you did a dialog for the root folder, but it actually doesn't close even after you selected a root folder. It only goes away after you closed and reopened the app

AbandonedCart, 2024-12-10:
It appears it isn't dismissing automatically, but also clicking outside the dialog dismisses it.

AbandonedCart, 2025-01-02:
The advanced section of settings now has an option to automatically enable skipping lock info. This still allows the ability to temporarily disable it before an individual write. That should cover being able to enable / disable it for a single write and also have it enabled / disabled by default.

closed 2025-01-11 by AbandonedCart