HiddenRamblings/TagMo #730: [Issue]: Writing to Amiibolink (Bluetooth) results in corrupt amiibo

LRFLEW, 2024-04-12:
### Issue Description

I recently bought an Amiibolink, and I'm trying to use it with TagMo. I've noticed that when I first upload a backup to the Amiibolink via bluetooth, it ends up in a state that the switch considers corrupt. I can get it to work by either using the amiibo reset option in the switch's settings or using Update in TagMo (with "Ignore Tag ID #" checked), and it then works fine (until writing another tag via bluetooth).

This is what the Switch shows when trying to scan the Amiibolink after writing to it:
[image]

I have the images from "View Hex Code" of both a Foomiibo and the Amiibolink scan after writing it, but I don't know if that is considered sharing "backups." There's also the possibility that it's just my device (I do get random errors occasionally from scanning it), but this issue is consistent, unlike the occasional errors I'm getting.

### Logcat Output

```shell
TagMo 4.1.7 (Play Release) #7204cf6cGoogle UPSIDE_DOWN_CAKE (14) - 12.00 GB RAM
--------- beginning of main04-12 00:15:15.487 29323 29323 E .eightbit:TagMo: Not starting debugger since process cannot load the jdwp agent.04-12 00:15:15.503 29323 29323 W ziparchive: Unable to open '/data/app/~~8qhe0fqd4l3WIEOCvOKgXA==/com.hiddenramblings.tagmo.eightbit-jRu3ZXLWMGb6BQYydB9uVw==/split_config.arm64_v8a.dm': No such file or directory04-12 00:15:15.503 29323 29323 W ziparchive: Unable to open '/data/app/~~8qhe0fqd4l3WIEOCvOKgXA==/com.hiddenramblings.tagmo.eightbit-jRu3ZXLWMGb6BQYydB9uVw==/split_config.arm64_v8a.dm': No such file or directory04-12 00:15:15.504 29323 29323 W .eightbit:TagMo: Entry not found04-12 00:15:15.504 29323 29323 W ziparchive: Unable to open '/data/app/~~8qhe0fqd4l3WIEOCvOKgXA==/com.hiddenramblings.tagmo.eightbit-jRu3ZXLWMGb6BQYydB9uVw==/split_config.en.dm': No such file or directory04-12 00:15:15.504 29323 29323 W ziparchive: Unable to open '/data/app/~~8qhe0fqd4l3WIEOCvOKgXA==/com.hiddenramblings.tagmo.eightbit-jRu3ZXLWMGb6BQYydB9uVw==/split_config.en.dm': No such file or directory04-12 00:15:15.504 29323 29323 W .eightbit:TagMo: Entry not found04-12 00:15:15.504 29323 29323 W ziparchive: Unable to open '/data/app/~~8qhe0fqd4l3WIEOCvOKgXA==/com.hiddenramblings.tagmo.eightbit-jRu3ZXLWMGb6BQYydB9uVw==/split_config.xxhdpi.dm': No such file or directory04-12 00:15:15.504 29323 29323 W ziparchive: Unable to open '/data/app/~~8qhe0fqd4l3WIEOCvOKgXA==/com.hiddenramblings.tagmo.eightbit-jRu3ZXLWMGb6BQYydB9uVw==/split_config.xxhdpi.dm': No such file or directory04-12 00:15:15.505 29323 29323 W .eightbit:TagMo: Entry not found04-12 00:15:15.632 29323 29323 W View    : requestLayout() improperly called by android.widget.FrameLayout{20bbf63 V.E..V... ......ID 0,0-1344,2601 #2} during layout: running second layout pass04-12 00:15:15.632 29323 29323 W View    : requestLayout() improperly called by androidx.recyclerview.widget.RecyclerView{ce2d760 IFED.V... ......I. 0,2613-1344,2613 #7f090109 app:id/foomiibo_list} during layout: running second layout pass04-12 00:15:15.632 29323 29323 W View    : requestLayout() improperly called by androidx.recyclerview.widget.RecyclerView{e752419 VFED.V... ......ID 0,0-1344,2598 #7f090082 app:id/browser_content} during layout: running second layout pass04-12 00:15:15.650 29323 29323 W TextView: onProvideContentCaptureStructure(): calling assumeLayout()04-12 00:15:18.223 29323 29323 W InputMethodManager: Ignoring hideSoftInputFromView() as view=androidx.appcompat.widget.AppCompatTextView{a76bd13 I.ED.V... ......I. 0,0-0,0 #7f0901bf app:id/np__numberpicker_input} is not served.04-12 00:15:18.255 29323 29323 W RecyclerView: No adapter attached; skipping layout04-12 00:15:18.265 29323 29323 W InputMethodManager: Ignoring hideSoftInputFromView() as view=androidx.appcompat.widget.AppCompatButton{a48750e VFED.VC.. ......I. 0,0-0,0 #7f09030f app:id/write_slot_count} is not served.04-12 00:15:18.266 29323 29323 W InputMethodManager: Ignoring hideSoftInputFromView() as view=androidx.appcompat.widget.AppCompatButton{da853c VFED.VC.. ......I. 0,0-0,0 #7f0900e7 app:id/erase_slot_count} is not served.04-12 00:15:18.581 29323 29323 W WindowOnBackDispatcher: sendCancelIfRunning: isInProgress=falsecallback=com.google.android.material.motion.MaterialBackOrchestrator$Api34BackCallbackDelegate$1@5512ad04-12 00:15:19.875 29323 29323 W WindowOnBackDispatcher: sendCancelIfRunning: isInProgress=falsecallback=android.app.Dialog$$ExternalSyntheticLambda2@fc75c9104-12 00:15:21.146 29323 29323 W RecyclerView: No adapter attached; skipping layout04-12 00:15:21.665 29323 29323 W RecyclerView: No adapter attached; skipping layout04-12 00:15:23.135 29323 29323 W RecyclerView: No adapter attached; skipping layout04-12 00:15:25.003 29323 29323 W RecyclerView: No adapter attached; skipping layout04-12 00:15:25.407 29323 29323 W RecyclerView: No adapter attached; skipping layout04-12 00:15:25.867 29323 29332 W JavaBinder: BinderProxy is being destroyed but the application did not call unlinkToDeath to unlink all of its death recipients beforehand.  Releasing leaked death recipient: com.google.android.play.core.internal.zzai04-12 00:15:27.846 29323 29323 W WindowOnBackDispatcher: sendCancelIfRunning: isInProgress=falsecallback=android.app.Dialog$$ExternalSyntheticLambda2@e53be2a04-12 00:15:28.117 29323 29323 W RecyclerView: No adapter attached; skipping layout04-12 00:16:06.409 29323 29323 W MaterialBackHelper: Must call startBackProgress() before updateBackProgress()04-12 00:16:06.496 29323 29323 W TextView: onProvideContentCaptureStructure(): calling assumeLayout()04-12 00:16:06.496 29323 29323 W TextView: onProvideContentCaptureStructure(): calling assumeLayout()04-12 00:16:06.497 29323 29323 W TextView: onProvideContentCaptureStructure(): calling assumeLayout()04-12 00:16:06.497 29323 29323 W TextView: onProvideContentCaptureStructure(): calling assumeLayout()04-12 00:16:06.497 29323 29323 W TextView: onProvideContentCaptureStructure(): calling assumeLayout()04-12 00:16:07.968 29323 29323 W WindowOnBackDispatcher: sendCancelIfRunning: isInProgress=falsecallback=android.widget.PopupWindow$PopupDecorView$$ExternalSyntheticLambda1@f669d6
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2024-04-12:
It sounds like the default app data is writing incorrectly. I'll look into what's going on there.

AbandonedCart, 2024-04-12:
It's interesting they would have the same issue, since they are completely unrelated. Our code is based on one of the Chinese applications that was circulating, along with logs from our Discord.

LRFLEW, 2024-04-13:
>It's interesting they would have the same issue, since they are completely unrelated. Our code is based on one of the Chinese applications that was circulating, along with logs from our Discord.

I found that other project from [this reddit comment](https://www.reddit.com/r/Amiibomb/comments/14ukr8y/comment/jrqkns1/), which indicated that it might have been a source of information for this project. I'm sure you know better than me whether these projects are related, but I'm sure they're based on at least similar reverse engineering work.

>What mode is your device set to when this happens? Does changing the mode act differently?

I'm not aware of any different "modes" on my device. If it does, then the (poorly translated) manual doesn't seem to mention it.

I saw you mentioned a Discord. I went ahead and joined (I'm @LRFLEW on there), so if you rather, we can try to troubleshoot this over there. I can't rule out a hardware problem just yet, so I might try testing some things on my own to see if there's a pattern to the errors.

AbandonedCart, 2024-04-13:
I only mentioned the difference because porting that site was suggested at one point, but it wasn't compatible and I didn't want to end up causing any confusion that what works for one will work for the other.

The devices should have three modes. Once allows the serial number to be changed every time you click, another changes it every time you scan, and a third shares a serial number among all the items written to it. I was curious about the mode because it could be a factor in why it's acting like that.

It's been a while since that was added, but I remember it being that the reset done in TagMo was based on the base code for Foomiibo, where a lot of sites and apps use a generic data set. It having the same results with two different methods means it may not be the actual data causing the issue.

LRFLEW, 2024-04-13:
Yeah, I don't know anything about any "modes", and can't figure out. I only see the one button, and it only seems to turn on the bluetooth. I know github issues are not really the place for this kind of tech support, so if you're willing to help explain how I'm supposed to use the device and switch "modes", it might be better to bring this conversation to Discord.

I do have an update, though. I was testing things, sending various data to the device and scanning it at different times, and noticed that the data is only "corrupted" when the device is on. If I disconnect the device, then wait for the pairing mode to time out, scanning the device returns the correct data. I was thinking *maybe* that meant the device was waiting for some sort of "confirmation" packet it wasn't getting to finalize the data. However, pressing the button to turn it back on returns the corruption. With that in mind, I'm starting to think this might be a hardware issue, where the bluetooth is interfering with the NFC.

AbandonedCart, 2024-07-19:
Closing this here, since it was moved to Discord.

closed 2024-07-19 by AbandonedCart