HiddenRamblings/TagMo #733: Key load error

acowsays, 2024-05-23:
### Issue Description

Unable to auto/manual import amitool keys.

Might be human error.

Requesting step by step instructions ^-^
Moo

### Logcat Output

```shell
N/a
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2024-05-23:
Unfortunately, there isn't enough information here to diagnose a cause and it isn't something I can reproduce in normal testing.

acowsays, 2024-05-23:
Understood, here is a video demonstrating my issue

https://github.com/HiddenRamblings/TagMo/assets/47616873/c0662380-97da-41da-a347-da262851f3df

AbandonedCart, 2024-05-23:
Seems I can’t play the video, but typically the logs provide the specific cause. That’s why they are required by the report.

AbandonedCart, 2024-05-24:
The video suddenly started working. The best I can guess from that is that the keys you have are invalid. You can also select both at the same time. That's not a huge shortcut, but every second counts.

acowsays, 2024-05-24:
Yeh that's totally  possible, i used these same keys for the 2.6.1 version of tagmo n it work.
So i just need to find n updated version of the keys. Hopefully i can find some, ill let you know if they work when i get the keys 🔑

AbandonedCart, 2024-05-26:
A few others have reported a bug that seems quite similar and included the logs. Assuming it was the same, this should be fixed in 9ea5c55. Let me know if it wasn't and I can open this issue back up.

closed 2024-05-26 by AbandonedCart