HiddenRamblings/TagMo #734: [Issue]: Can not scroll in Bluetooth window.

ottoxgam, 2024-07-16:
### Issue Description

I am unable to scroll through the enumerated Bluetooth devices while in the Bluetooth window. This issue is not affected by portrait vs landscape. 

https://imgur.com/a/A3kOwZV

### Logcat Output

```shell
ImeTracker
com.hiddenramblings.tagmo.eightbit:TagMo:ff916a96: onRequestHide at ORIGIN_CLIENT_HIDE_SOFT_INPUT reason HIDE_SOFT_INPUT_FROM_VIEW

ImeTracker
com.hiddenramblings.tagmo.eightbit:TagMo:ff916a96: onFailed at PHASE_CLIENT_VIEW_SERVED

ImeTracker
com.hiddenramblings.tagmo.eightbit:TagMo:b954826f: onRequestHide at ORIGIN_CLIENT_HIDE_SOFT_INPUT reason HIDE_SOFT_INPUT_FROM_VIEW
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2024-07-16:
Google Play or GitHub?

ottoxgam, 2024-07-16:
TagMo 4.2.2 (GitHub Release) #d1c1d16

AbandonedCart, 2024-07-19:
This should be fixed in the newest build.

ottoxgam, 2024-07-20:
Thank you! Works as expected!

closed 2024-07-20 by ottoxgam

AbandonedCart, 2024-07-20:
Great! Thanks for the report.