HiddenRamblings/TagMo #735: New amiibo not appearing with photos

AbandonedCart, 2024-09-05:
Continuation of https://github.com/N3evin/AmiiboAPI/issues/181

N3evin, 2024-09-05:
Are the other data populated? And just images?

AbandonedCart, 2024-09-05:
Images are pulled from the API. There is a local database copy that’s superseded by the API if the API is newer.

N3evin, 2024-09-05:
Anyway to debug the image url you are getting? From the APIs I can see that the image url is working.

AbandonedCart, 2024-09-05:
I’m wondering if the issue is that it was synced when I had 3 of the image names wrong. It lines up with the issue. I think clearing the cache in TagMo would fix it.

N3evin, 2024-09-05:
That could be it. Give that a try.

AbandonedCart, 2024-09-05:
@Brickleberry19 Try going into settings and using the option to clear the database cache.

Brickleberry19, 2024-09-05:
> @Brickleberry19 Try going into settings and using the option to clear the database cache.

Yea i did it and it doesnt fix it

AbandonedCart, 2024-09-05:
You can also try refreshing the database from the same menu (it's labeled sync) or wait for the new update to do it automatically.

N3evin, 2024-09-05:
I decided to install tagmo. Same issue even with sync. Installed from play store. Shall try with the apk build

N3evin, 2024-09-05:
Mind telling me the key id? I believe that's the issue. The one I downloaded had 08020000 instead of 08020200 for Marie. Fake bin I believe

Brickleberry19, 2024-09-05:
> @Brickleberry19 mind telling me the key id? I believe that's the issue. The one I downloaded had 08020000 instead of 08020200 for Marie. Fake bin is going around.

Marie
0802000004370402

Callie
0801000004360402

Pearl 
0803000004380402

Marina
0804000004390402

N3evin, 2024-09-05:
If that's the case the data that @AbandonedCart provided to amiiboapi is spoof data.

N3evin, 2024-09-05:
I will update it tomorrow morning.

AbandonedCart, 2024-09-05:
> If that's the case the data that @AbandonedCart provided to amiiboapi is spoof data.

Or a typo...

AbandonedCart, 2024-09-05:
@Brickleberry19 And you're absolutely sure they are entered correctly in your post? This is why I asked for verification and didn't want to move forward with the PR until it was confirmed.

AbandonedCart, 2024-09-05:
I think whoever sent them to me typed them manually and may have introduced the typos by mistake.

AbandonedCart, 2024-09-05:
They should now be fixed in TagMo. Next build will have an option in the share / export to copy the ID. PR submitted to @N3evin to fix the API. It's still way ahead of the usual schedule for new additions, so this was just a friendly reminder of how every single digit counts.

closed 2024-09-05 by AbandonedCart

AbandonedCart, 2024-09-05:
Copy option still needs work. Seems I have it exporting the serial. That's not useful.

AbandonedCart, 2024-09-05:
Marie - 0802000004370402

Brickleberry19, 2024-09-05:
Thank you for this Update , Its a great Feature

Brickleberry19, 2024-09-05:
> Copy option still needs work. Seems I have it exporting the serial. That's not useful.
> 
> @Brickleberry19 Why do I keep forgetting to add you as co-author? You are a vital part of the process and your help is greatly appreciated.

I don't really care if you add me as a co-author or not🤷🏻‍♂️😅

AbandonedCart, 2024-09-05:
Yeah, but it’s deserved. You were doing twice the work you should because TagMo lacked such an obvious feature.

Brickleberry19, 2024-09-05:
Taking a screenshot and copying the ID from it is not that difficult and takes less than 10 seconds

Brickleberry19, 2024-09-05:
Thank you anyway for adding the option and for continuing to work on the app. You are doing a good job