HiddenRamblings/TagMo #736: App doesn't properly display on Pixel 7

Theblackcat98, 2024-12-20:
### Issue Description

When I open the app, it expands beyond the notification bar and bottom bar. Making it so some buttons and functions are hidden or unusable.
[image]
[image]

### Logcat Output

```shell
TagMo 4.3.0 (Play Release) #13397a3b
Google VANILLA_ICE_CREAM (15) - 8.00 GB RAM

--------- beginning of main
12-20 15:35:10.163 27159 27159 E .eightbit:TagMo: Not starting debugger since process cannot load the jdwp agent.
12-20 15:35:10.189 27159 27159 W ziparchive: Unable to open '/data/app/~~fCIDknv4436rX5mQ81bESw==/com.hiddenramblings.tagmo.eightbit-OP3AaCcwUOYtNlbJ9TN8uQ==/split_config.arm64_v8a.dm': No such file or directory
12-20 15:35:10.189 27159 27159 W ziparchive: Unable to open '/data/app/~~fCIDknv4436rX5mQ81bESw==/com.hiddenramblings.tagmo.eightbit-OP3AaCcwUOYtNlbJ9TN8uQ==/split_config.arm64_v8a.dm': No such file or directory
12-20 15:35:10.189 27159 27159 W .eightbit:TagMo: Entry not found
12-20 15:35:10.190 27159 27159 W ziparchive: Unable to open '/data/app/~~fCIDknv4436rX5mQ81bESw==/com.hiddenramblings.tagmo.eightbit-OP3AaCcwUOYtNlbJ9TN8uQ==/split_config.en.dm': No such file or directory
12-20 15:35:10.190 27159 27159 W ziparchive: Unable to open '/data/app/~~fCIDknv4436rX5mQ81bESw==/com.hiddenramblings.tagmo.eightbit-OP3AaCcwUOYtNlbJ9TN8uQ==/split_config.en.dm': No such file or directory
12-20 15:35:10.190 27159 27159 W .eightbit:TagMo: Entry not found
12-20 15:35:10.190 27159 27159 W ziparchive: Unable to open '/data/app/~~fCIDknv4436rX5mQ81bESw==/com.hiddenramblings.tagmo.eightbit-OP3AaCcwUOYtNlbJ9TN8uQ==/split_config.xxhdpi.dm': No such file or directory
12-20 15:35:10.190 27159 27159 W ziparchive: Unable to open '/data/app/~~fCIDknv4436rX5mQ81bESw==/com.hiddenramblings.tagmo.eightbit-OP3AaCcwUOYtNlbJ9TN8uQ==/split_config.xxhdpi.dm': No such file or directory
12-20 15:35:10.190 27159 27159 W .eightbit:TagMo: Entry not found
12-20 15:35:10.233 27159 27159 W HWUI    : Image decoding logging dropped!
12-20 15:35:10.243 27159 27159 W HWUI    : Image decoding logging dropped!
12-20 15:35:10.248 27159 27159 W HWUI    : Image decoding logging dropped!
12-20 15:35:10.277 27159 27159 W ResourcesCompat: Failed to inflate ColorStateList, leaving it to the framework
12-20 15:35:10.277 27159 27159 W ResourcesCompat: java.lang.UnsupportedOperationException: Failed to resolve attribute at index 0: TypedValue{t=0x2/d=0x7f040114 a=-1}, theme={InheritanceMap=[id=0x7f130258com.hiddenramblings.tagmo.eightbit:style/Theme.Design.Light.BottomSheetDialog, id=0x7f13024fcom.hiddenramblings.tagmo.eightbit:style/Theme.AppCompat.Light.Dialog, id=0x7f130056com.hiddenramblings.tagmo.eightbit:style/Base.Theme.AppCompat.Light.Dialog, id=0x7f1300a9com.hiddenramblings.tagmo.eightbit:style/Base.V21.Theme.AppCompat.Light.Dialog, id=0x7f1300c2com.hiddenramblings.tagmo.eightbit:style/Base.V7.Theme.AppCompat.Light.Dialog, id=0x7f130054com.hiddenramblings.tagmo.eightbit:style/Base.Theme.AppCompat.Light, id=0x7f1300becom.hiddenramblings.tagmo.eightbit:style/Base.V28.Theme.AppCompat.Light, id=0x7f1300bbcom.hiddenramblings.tagmo.eightbit:style/Base.V26.Theme.AppCompat.Light, id=0x7f1300b5com.hiddenramblings.tagmo.eightbit:style/Base.V23.Theme.AppCompat.Light, id=0x7f1300b3com.hiddenramblings.tagmo.eightbit:style/Base.V22.Theme.AppCompat.Light, id=0x7f1300a8com.hiddenramblings.tagmo.eightbit:style/Base.V21.Theme.AppCompat.Light, id=0x7f1300c1com.hiddenramblings.tagmo.eightbit:style/Base.V7.Theme.AppCompat.Light, id=0x7f130141com.hiddenramblings.tagmo.eightbit:style/Platform.AppCompat.Light, id=0x7f13014ccom.hiddenramblings.tagmo.eightbit:style/Platform.V25.AppCompat.Light, id=0x1030241android:style/Theme.Material.Light.NoActionBar, id=0x1030237android:style/Theme.Material.Light, id=0x103000candroid:style/Theme.Light, id=0x1030005android:style/Theme], Themes=[com.hiddenramblings.tagmo.eightbit:style/Theme.Design.Light.BottomSheetDialog, forced, com.hiddenramblings.tagmo.eightbit:style/PreferenceThemeOverlay.v14.Material, not forced, com.hiddenramblings.tagmo.eightbit:style/AppTheme, forced, com.hiddenramblings.tagmo.eightbit:style/Theme.AppCompat.Empty, forced, android:style/Theme.DeviceDefault.Light.DarkActionBar, forced]}
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at android.content.res.TypedArray.getColor(TypedArray.java:536)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.core.content.res.ColorStateListInflaterCompat.inflate(ColorStateListInflaterCompat.java:159)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.core.content.res.ColorStateListInflaterCompat.createFromXmlInner(ColorStateListInflaterCompat.java:124)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.core.content.res.ColorStateListInflaterCompat.createFromXml(ColorStateListInflaterCompat.java:103)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.core.content.res.ResourcesCompat.inflateColorStateList(ResourcesCompat.java:263)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.core.content.res.ResourcesCompat.getColorStateList(ResourcesCompat.java:237)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.core.content.ContextCompat.getColorStateList(ContextCompat.java:519)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.appcompat.content.res.AppCompatResources.getColorStateList(AppCompatResources.java:46)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.appcompat.widget.TintTypedArray.getColorStateList(TintTypedArray.java:178)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.appcompat.widget.AppCompatBackgroundHelper.loadFromAttributes(AppCompatBackgroundHelper.java:66)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.appcompat.widget.AppCompatButton.<init>(AppCompatButton.java:86)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.appcompat.widget.AppCompatButton.<init>(AppCompatButton.java:76)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.appcompat.app.AppCompatViewInflater.createButton(AppCompatViewInflater.java:238)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.appcompat.app.AppCompatViewInflater.createView(AppCompatViewInflater.java:156)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.appcompat.app.AppCompatDelegateImpl.createView(AppCompatDelegateImpl.java:1654)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.appcompat.app.AppCompatDelegateImpl.onCreateView(AppCompatDelegateImpl.java:1705)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at android.view.LayoutInflater$FactoryMerger.onCreateView(LayoutInflater.java:225)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at android.view.LayoutInflater.tryCreateView(LayoutInflater.java:949)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:885)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:849)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at android.view.LayoutInflater.rInflate(LayoutInflater.java:1011)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:972)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at android.view.LayoutInflater.rInflate(LayoutInflater.java:1014)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:972)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at android.view.LayoutInflater.rInflate(LayoutInflater.java:1014)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:972)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at android.view.LayoutInflater.inflate(LayoutInflater.java:570)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at android.view.LayoutInflater.inflate(LayoutInflater.java:462)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at com.hiddenramblings.tagmo.fragment.FittedSheets.onCreateView(FittedSheets.kt:48)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.fragment.app.Fragment.performCreateView(Fragment.java:3119)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.fragment.app.DialogFragment.performCreateView(DialogFragment.java:770)
12-20 15:35:10.277 27159 27159 W ResourcesCompat: 	at androidx.fragment.app.FragmentStateManager.createView(FragmentStateManager.java:577)
[... cut ...]
12-20 15:35:20.436 27159 27159 W HWUI    : Image decoding logging dropped!
12-20 15:35:20.436 27159 27159 W HWUI    : Image decoding logging dropped!
12-20 15:35:20.447 27159 27163 W .eightbit:TagMo: Cleared Reference was only reachable from finalizer (only reported once)
12-20 15:35:23.743 27159 27159 W WindowOnBackDispatcher: sendCancelIfRunning: isInProgress=false callback=android.widget.PopupWindow$PopupDecorView$$ExternalSyntheticLambda1@59c373b
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2024-12-21:
Interesting. That's a system component. Is that a beta Android version?

Theblackcat98, 2024-12-21:
> Interesting. That's a system component. Is that a beta Android version?

Dang, yeah I forgot I'm running the latest Android 15 beta. I haven't had issues like this in the past, but the beta could definitely be involved.
Any other info you need that I can provide?

AbandonedCart, 2024-12-21:
I believe the issue is that Android breaks the status and navigation bars in Android 15.

Theblackcat98, 2024-12-21:
Found this:

[Android 15 Status Bar size increase](https://9to5google.com/2024/10/23/pixel-9-android-15-status-bar/#:~:text=With%20Android%2015%20QPR1%20Beta,Pixel%20Launcher%20homescreen%20moves%20down.&text=Meanwhile%2C%20things%20look%20much%20better,9%20and%209%20Pro%20XL.)

AbandonedCart, 2024-12-21:
The actual size doesn't matter, since it's registered as an inset. Android 15 removed the app commands to theme the system bars and, in doing so, also made the UI "edge to edge" but continued to register the size of the bars as insets.

Fixed in 6d94dd5

Theblackcat98, 2024-12-21:
I see! I don't know mucha about development, but I'm glad I was able to relay the correct information!
Also, with the latest commit the app seems to load and work as expected. I'm going to keep testing it.
[image]

AbandonedCart, 2024-12-21:
https://developer.android.com/about/versions/15/behavior-changes-15#ux

closed 2024-12-21 by AbandonedCart