HiddenRamblings/TagMo #738: Tagmo issue

buwnga, 2025-01-19:
### Issue Description

The error is

TagMo 4.3.3 (Play Release) #87054f42
motorola UPSIDE_DOWN_CAKE (14) - 14.00 GB RAM

java.lang.Exception: NTAG215 required, found NfcA!
	at com.hiddenramblings.tagmo.nfctech.NfcActivity.onMifareUltralight(NfcActivity.kt:515)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity.access$onMifareUltralight(NfcActivity.kt:52)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onTagDiscovered$2.invokeSuspend(NfcActivity.kt:683)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onTagDiscovered$2.invoke(NfcActivity.kt:0)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onTagDiscovered$2.invoke(NfcActivity.kt:0)
	at kotlinx.coroutines.intrinsics.UndispatchedKt.startUndispatchedOrReturn(Undispatched.kt:42)
	at kotlinx.coroutines.BuildersKt__Builders_commonKt.withContext(Builders.common.kt:164)
	at kotlinx.coroutines.BuildersKt.withContext(Unknown Source:1)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity.onTagDiscovered(NfcActivity.kt:675)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity.access$onTagDiscovered(NfcActivity.kt:52)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onNewIntent$1.invokeSuspend(NfcActivity.kt:825)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:101)
	at kotlinx.coroutines.internal.LimitedDispatcher$Worker.run(LimitedDispatcher.kt:113)
	at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:89)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:589)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:823)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:720)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:707)

### Logcat Output

```shell
TagMo 4.3.3 (Play Release) #87054f42
motorola UPSIDE_DOWN_CAKE (14) - 14.00 GB RAM

java.lang.Exception: NTAG215 required, found NfcA!
	at com.hiddenramblings.tagmo.nfctech.NfcActivity.onMifareUltralight(NfcActivity.kt:515)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity.access$onMifareUltralight(NfcActivity.kt:52)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onTagDiscovered$2.invokeSuspend(NfcActivity.kt:683)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onTagDiscovered$2.invoke(NfcActivity.kt:0)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onTagDiscovered$2.invoke(NfcActivity.kt:0)
	at kotlinx.coroutines.intrinsics.UndispatchedKt.startUndispatchedOrReturn(Undispatched.kt:42)
	at kotlinx.coroutines.BuildersKt__Builders_commonKt.withContext(Builders.common.kt:164)
	at kotlinx.coroutines.BuildersKt.withContext(Unknown Source:1)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity.onTagDiscovered(NfcActivity.kt:675)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity.access$onTagDiscovered(NfcActivity.kt:52)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onNewIntent$1.invokeSuspend(NfcActivity.kt:825)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:101)
	at kotlinx.coroutines.internal.LimitedDispatcher$Worker.run(LimitedDispatcher.kt:113)
	at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:89)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:589)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:823)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:720)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:707)
```

### Usage Requirements

- [x] I have read and understood the usage requirements for TagMo

AbandonedCart, 2025-01-19:
Do you have the correct tags?

buwnga, 2025-01-19:
I have Ntag 215

buwnga, 2025-01-19:
Sticker

AnimeWaifuGirl, 2025-11-11:
its not the tag the data are recognized as blank tag