HiddenRamblings/TagMo #739: [Issue]: Scanning mii qr code from 3ds is not workingd

Sorunome, 2025-04-10:
### Issue Description

When trying to scan a Mii QR code generated from the 3DS it gives "mac check in CCM failed"

Soru thinks this is likely because the 3DS does not implement aes-ccm correctly in hardware, as for the mac calculation it assumes that the buffer is a multiple of 0x10 bytes large. This is not the case for miis, as there the buffer is 0x68 bytes large (0x08 bytes of the 0x70 byte buffer are used for the nonce, instead of the ciphertext). Sadly soru can't seem to find the documentation for that in the 3dbrew wiki for that currently.

So, as a TL;DR: AES-CCM needs nopadding for the ciphertext but it needs padding for the verification of the mac.

When looking into tagmo's source code, it is implemented here: https://github.com/HiddenRamblings/TagMo/blob/8f2a1ff714752e1d770ecaed133ea33db3d2c8e3/app/src/main/java/com/hiddenramblings/tagmo/qrcode/QRCodeScanner.kt#L183-L195 and we can see that everything is just nopadding.

In sorus implementation of encrypting / decrypting miis, she decrypts without verifying the mac, then encrypts again, with padded 0x8 0-bytes, and then verifies the re-encrypted mac. You can find this here, if you need to: https://gitlab.com/3ds-netpass/netpass-server/-/blob/main/src/types/mii.rs?ref_type=heads#L307

### Logcat Output

```shell
No usb cable for phone atm, can fill in if needed
```

### Usage Requirements

- [x] I have read and understood the usage requirements for TagMo

AbandonedCart, 2025-04-11:
Your explanation is a bit lost in translation, but what QR code are you using to test?

Sorunome, 2025-04-11:
Ah sorry, this is in the QR code toolkit when trying to scan a qr code that contains a mii that was generated on the 3DS

AbandonedCart, 2025-04-11:
Did you not save it anywhere?

Sorunome, 2025-04-11:
Ah sorry, soru can be a bit dense at times ^^

[image]

reewhy, 2025-06-14:
Is this fixed yet?

closed 2026-01-30 by AbandonedCart